volkszaehler/vzlogger #416: Simplification of equality comparison for ReadingIdentifer

claashk, 2020-01-17:
While I was at it, I made some changes (improvements?) to the ReadingIdentifier Code. The new version of the comparison operator does no longer rely on the knowledge of all derived classes. 

See 6c701f86dfbdf5a60c4d47b99b63d3dbddbf980a for the proposed change. Note that the build only fails because of the relative paths used in the tests, which are fixed by #414 

If you think this would is an improvement, too, I'll gladly create a pull request.

andig, 2020-03-26:
/cc @mbehr1

andig, 2020-03-26:
Closing here to avoid noise, please feel free to open a PR if you think this is worthwhile.

closed 2020-03-26 by andig

claashk, 2020-03-28:
Could you define noise ? I proposed the linked changes for discussion. I got feedback from @mbehr1, that he is not sure whether to incorporate these changes, since he is not convinced, that the test coverage is sufficient. Obviously I would like to avoid creating a pull request, that will not be included. So if more tests are required, I could add them, if someone familiar with the test environment would be so kind to provide a hint, where these should be implemented. 

I was under the impression, that there was a certain interest to polish up the existing code base. Is that a misunderstanding on my part ?

andig, 2020-03-28:
Sorry, I won‘t discuss with you on this basis. Apparently pointing to a commit didn‘t get traction. As I wrote- please open a PR if you want. I hooe that gets more attention.

claashk, 2020-03-28:
I feel this has been a constructive discussion with @mbehr1 so far, even though you may consider it noise. I admit, maybe we should have discussed in this issue board instead of using the commit.

Do I understand you correctly: A PR without modification of the unit tests is fine for you ?

mbehr1, 2020-03-28:
yep, please create a PR based on that commit.