volkszaehler/vzlogger #419: Improve MQTT Documentation

claashk, 2020-03-26:
From discussion on PR #418 

The documentation related to MQTT should be improved. Especially channel naming conventions and the general concept would be helpful.

andig, 2020-03-26:
Please feel free to do so. The wiki is open for changes.

closed 2020-03-26 by andig

claashk, 2020-03-28:
After I moved the discussion to the [mailing list (german)](https://lists.volkszaehler.org/pipermail/volkszaehler-dev/2020-March/006557.html) as you suggested, my impression was that the consensus was as follows: the initial MQTT article should be created by a core maintainer, who is familiar with the implementation and the wiki structure. Once it is clear, where to document the MQTT functionality and how the content should be structured, others (including me) would have a much easier time filling the gaps (if any). 

When you close the issue, does that mean the documentation has already been created or does it imply, that there will be no support from the core maintainers ?

andig, 2020-03-28:
We are an open source project with limited ressources. Everything is done on voluntary time. If you want to help you‘re welcome. If not I do not see anybody else picking up the task and hence closed the issue.

claashk, 2020-03-28:
I think we agree, that everybody here works in his/her free time. If your intention is to make my contributions feel welcome, I can only respond that it is not working so well. I want to explicitly state, that my discussions with other maintainers on the project have been very constructive and supporting so far. I tried to (hopefully kindly) ask for guidance, where and how to document something. Someone on the mailing list picked up on the topic and offered support. If this does not happen today, tomorrow or this year, that is of course fine. I do not understand, why you spend your no-doubt valuable free time with these kind of discussions, which do not seem to pursue a constructive solution ?

andig, 2020-03-28:
Deleted off-topic messages. For discussions refer to the mailing list.