volkszaehler/vzlogger #420: Change MQTT channel IDs to UUIDs

claashk, 2020-03-26:
It would seem beneficial to use the UUID as channel names for MQTT, too. As far as I understand, all MQTT channel names change if a channel is inserted into the config file anywhere but at the bottom. Of course the user application could parse the config file and map UUIDs to channel index, but first I am not sure whether user code should have read permission to the config at all and secondly this endeavor is not made easier by the fact, that the default config file is not valid json due to the comments and thus cannot be parsed by strict json parsers. In order to avoid breaking existing code, this behavior could be triggered by an option in the config file.

andig, 2020-03-26:
Agreed. Instead of uuids or in addition it might also be nice to use device/ reading type names.

r00t-, 2021-03-10:
i would suggest to NOT use the uuids for this (and actually restrict their use to the volkszaehler.org middleware api that uses them.
i think the only use-case where the UUIDs are useful is when using the public demo.volkszaehler.org server, where the cryptic un-guessable UUIDs are used as a replacement for authentication.
the influxdb interface also already implements it's own non-uuid identifiers.

claashk, 2021-04-07:
I agree that UUIDs are most likely not the best way to pursue this, but some unique naming scheme would be nice. Eventually the UUIDs could be created from these names instead ? In any case a single source of truth for InfluxDB, volkszaehler web interface and MQTT still seems like a good idea to me. What do you think ?

Zugschlus, 2021-12-15:
How about having a configuration option "mqtt_subtopic" in the meter stanza of the configuration? In the absence of that option, the channel number could still be used, allowing backwards compatibility and a flexible migration path.

abutscher, 2023-12-10:
> How about having a configuration option "mqtt_subtopic" in the meter stanza of the configuration? In the absence of that option, the channel number could still be used, allowing backwards compatibility and a flexible migration path.

@Zugschlus I like that proposal. Actually that is exactly what I was looking for in the current MQTT configuration and (obviously) did not find it there. Is there already any work going on on this topic? If yes, I'd be very interested and also willing to contribute ;-).