volkszaehler/vzlogger #421: Always use json for MQTT messages

claashk, 2020-03-26:
I found it quite surprising, that changing the option "timestamp" in the config changes the message format in MQTT from json to a raw binary string. This means users have to adapt code relying on a certain MQTT message format, whenever this setting is modified. Would it be possible to use json all the time ?

andig, 2020-03-26:
There was a reason for doing so. You might check the pr changing it or the mailing list at that time.

claashk, 2020-03-28:
I found PR #371, which seems to address the time stamps. I could not find a rationale there, why json is not used all the time, independent of the message content/timestamp setting. Did I overlook something ? 

In case this is hidden the depths of the mailing list archives: Could I ask you to summarise the essence of the discussion here ? This would probably be a huge help to others, trying to understand the decision making process in the future, especially since the discussion on the mailing list is mostly in German.

J-A-U, 2020-03-28:
Erster Ansatz wäre nach dem Datum zu schauen, wenn es eine Diskussion dazu gab dann wird die wohl um den Zeitraum liegen.

Aber da ist nix weiter.

andig, 2020-03-28:
I don‘t see any reason to always use json. It complicates things and is a breaking change. On the other side, to transport 2 values- timestamp and data- some structure needed and json chosen. Please clarify what problem this is causing for you.

claashk, 2020-03-28:
Yes it is a breaking change, that's why I would like to discuss options before implementing. It could be made 'less breaking', by adding a configuration option. I am not sure it complicates things, since json is already implemented as one possible protocol, so in a way it would simplify things by removing one protocol.

The problem is, that changing between "with timestamps" and "no timestamps" changes the underlying data format. Thus applications receiving the MQTT messages have to adapt, whenever the config option is changed. If only the "timestamp" message was missing from the json dictionary, that would be easy to check in application code. However, a change of message format can have more severe consequences, especially when using something like crossbar.io, which translates the packages from MQTT to WAMP. This is especially true, when using crossbar.io as bridge in [native mode](https://crossbar.io/docs/MQTT-Broker/) with a static json serializer.

andig, 2020-03-28:
I feel this has been clarified. Having simple values is helpful for many clients and we'll not break that. Feel free to open a PR (which legacy fallback) that renames the timestamp options to something for sensible.

closed 2020-03-28 by andig