volkszaehler/vzlogger #426: Assign the same timestamp to measurements from a single transmission

workoft, 2020-04-29:
Meters with d0 interface will typically transmit several values within one burst. vzlogger does not group these bursts, but instead treats each line individually. This causes values from a single burst to have slightly different timestamps, which in turn is ugly if you are trying to do calculations with these measurements.
Is there a way to assign a single timestamp to all values of one burst, such as by re-using timestamps that aren't older than a specified time? Or by specifying the first channel that is transmitted as the timestamp channel and having all other channels just copy the last timestamp from that channel?

This is related to issue #417 - the user is asking to group measurements in the database (but also from different meters). I'm asking how to assign the same timestamp to measurements from one transmission burst.

r00t-, 2021-03-10:
@workoft:
this issue should only occur when you enable `use_local_time`,
your meter is not sending timestamps?

also, your interpretation does not seem quite true.
vzlogger reads a whole "burst" from the meter, and then processes it:
https://github.com/volkszaehler/vzlogger/blob/1a6e10b6d8d69282f3df8152c35ce1b1f19baad7/src/protocols/MeterSML.cpp#L279
but does attach a new microsecod-precision timestamp to each one (when `use_local_time` is set):
https://github.com/volkszaehler/vzlogger/blob/1a6e10b6d8d69282f3df8152c35ce1b1f19baad7/src/protocols/MeterSML.cpp#L333

actually the differences between those timestamps should be really, really tiny, as it's differences are only due to the execution time of the C code.

but the fix should be trivial, just generate a timestamp in after `sml_file_parse()` and then pass it into `_parse()` for all readings.
or just overwrite `rds[m]->time()` in the caller, to avoid changing  `_parse()`.

r00t-, 2021-03-11:
@workoft: can you test the fix in #455

r00t-, 2021-04-14:
@workoft: ping