volkszaehler/vzlogger #427: vzlogger crashes when MQTT is switched on (SEGV)

ahemwe, 2020-06-29:
I've been running vzloger for about one year without any problems.
After moving the middleware to an other Raspi and activating MQTT, vzlogger crashes roughly every day or after some hours.
```
pi@vz:~ $ sudo systemctl status vzlogger
● vzlogger.service - vzlogger
   Loaded: loaded (/etc/systemd/system/vzlogger.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Mon 2020-06-29 18:12:29 CEST; 3h 18min ago
  Process: 635 ExecStart=/usr/local/bin/vzlogger -c /etc/vzlogger.conf (code=killed, signal=SEGV)
 Main PID: 635 (code=killed, signal=SEGV)

Jun 29 14:05:45 vz systemd[1]: Started vzlogger.
Jun 29 18:12:29 vz systemd[1]: vzlogger.service: Main process exited, code=killed, status=11/SEGV
Jun 29 18:12:29 vz systemd[1]: vzlogger.service: Failed with result 'signal'.
pi@vz:~ $
```
I've investigated the issue and found, that it works well for weeks when swichting off MQTT, but crashes with SEGV when MQTT is switched on.

Hardware:
- Raspi Zero
- Sensor: https://wiki.volkszaehler.org/howto/simpler_ir_leser

Software:
- MQTT Broker: mosquitto

Already done:
- Updated vzlogger to lastest git version (Commit 8d06176360fec9455666b41da425c14003839054) and rebuild
- Installed an other sd card
- checked power supply

vzlogger shows no error message, even on debug vebosity 15.
Does anybody give me a hint for further investigations? I'd like to use the mqtt feature for some home automation.

vzlogger.conf:
[vzlogger.conf.txt](https://github.com/volkszaehler/vzlogger/files/4847670/vzlogger.conf.txt)

mbehr1, 2020-07-18:
can you generate a coredump/backtrace? (btw: german language possible if preferred)

ahemwe, 2020-07-18:
Sure.
Could you please explain how to do so? Thanks for looking into it.

andig, 2020-07-18:
See https://github.com/volkszaehler/vzlogger/issues/428#issuecomment-653846695 for how to

ahemwe, 2020-07-18:
Sorry for bothering, but do I have to recompile to get debuging symbols into the binary?
`bt ` and `bt full` give both 
_No stack_

mbehr1, 2020-07-18:
should not be needed. so you have a coredump and this leads to "no stack" if loaded via gbd <full path to vzlogger binary> <your core dump file> ?
Can you attach the coredump?

ahemwe, 2020-07-21:
Please find one coredump attached. Thanks.
I've put it into a gz cause github didn't accept lz4-files.

[core.vzlogger.0.1a5b6818f6eb47d5b1cab7924e81a5ff.5362.1595297141000000.lz4.gz](https://github.com/volkszaehler/vzlogger/files/4955986/core.vzlogger.0.1a5b6818f6eb47d5b1cab7924e81a5ff.5362.1595297141000000.lz4.gz)

mbehr1, 2020-07-22:
thx. can you create the backtrace as well (or attach the binary vzlogger)?

ahemwe, 2020-07-22:
Here we go:
[vzlogger.gz](https://github.com/volkszaehler/vzlogger/files/4962335/vzlogger.gz)

mbehr1, 2020-07-23:
Core was generated by `/usr/local/bin/vzlogger -c /etc/vzlogger.conf'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0010fe1c in Reading::deleted (this=0x8) at /home/pi/vzlogger/include/Reading.hpp:149
149	/home/pi/vzlogger/include/Reading.hpp: No such file or directory.
[Current thread is 1 (LWP 5365)]
(gdb) bt
#0  0x0010fe1c in Reading::deleted (this=0x8) at /home/pi/vzlogger/include/Reading.hpp:149
#1  0x0011b5e8 in reading_thread (arg=0x1d431a8) at /home/pi/vzlogger/src/threads.cpp:148
#2  0xb6e4f494 in ?? ()

mbehr1, 2020-07-23:
(output of gdb-multiarch vzlogger core...)

mbehr1, 2020-07-23:
gdb) f 1
#1  0x0011b5e8 in reading_thread (arg=0x1d431a8) at /home/pi/vzlogger/src/threads.cpp:148
148	/home/pi/vzlogger/src/threads.cpp: No such file or directory.
(gdb) info frame
Stack level 1, frame at 0xb4ed4be8:
 pc = 0x11b5e8 in reading_thread (/home/pi/vzlogger/src/threads.cpp:148); saved pc = 0xb6e4f494
 called by frame at 0xb4ed4be8, caller of frame at 0xb4ed49a8
 source language c++.
 Arglist at 0xb4ed4be4, args: arg=0x1d431a8
 Locals at 0xb4ed4be4, Previous frame's sp is 0xb4ed4be8
 Saved registers:
  r4 at 0xb4ed4bd8, r5 at 0xb4ed4bdc, r11 at 0xb4ed4be0, lr at 0xb4ed4be4, s16 at 0xb4ed4bd0, s17 at 0xb4ed4bd4
(gdb) info locals
r = <error reading variable>
buf = {<std::tr1::__shared_ptr<Buffer, (__gnu_cxx::_Lock_policy)1>> = {_M_ptr = 0x1d40200, _M_refcount = {_M_pi = 0x1d423d0}}, <No data fields>}
it = {_M_node = 0x0}
ch = {_M_current = 0x1d445d0}
mapping = 0x1d431a8
mtr = {<std::tr1::__shared_ptr<Meter, (__gnu_cxx::_Lock_policy)1>> = {_M_ptr = 0x1d3d8f8, _M_refcount = {_M_pi = 0x1d41150}}, <No data fields>}
aggIntEnd = 1595167023
details = 0x2888d0 <protocols+120>
n = 2
rds = {<std::_Vector_base<Reading, std::allocator<Reading> >> = {_M_impl = {<std::allocator<Reading>> = {<__gnu_cxx::new_allocator<Reading>> = {<No data fields>}, <No data fields>}, _M_start = 0xb2b00ac8, 
      _M_finish = 0xb2b00ec8, _M_end_of_storage = 0xb2b00ec8}}, <No data fields>}

andig, 2020-07-23:
Du brabbelst 😂

mbehr1, 2020-07-23:
@andig sonst vergesse ich immer wie man die coredumps aufruft :-)

mbehr1, 2020-07-23:
@ahemwe can you try the fix from PR#429 ?

ahemwe, 2020-07-23:
Sure. I've just pulled your commit and ran the make.
I'll let you know if everything runs or crashes. I've just stopped it after 21 hours without crash, but usually it crashed earlier.

ahemwe, 2020-07-25:
I suggest consider this issue as fixed since it's now running for 1 d 8 h without crash. 
Thanks a lot.

mbehr1, 2020-07-25:
perfect. merged #429. closing this one.

closed 2020-07-25 by mbehr1