volkszaehler/vzlogger #428: vzlogger crash after some Minutes Status=11/SEGV

Morphi08, 2020-07-04:
Hello,
i install e new Image on my Raspi 3 from Volkszahler; after that I updated all programs and edited my config.
All thinks would be Ok, but after some time, maybe 20 Minutes, the vzlogger deamon crash; in the Logfile is nothing unnormal, but in the syslog there is this statement:
`
Jul  4 18:05:59 raspberrypi systemd[1]: vzlogger.service: Main process exited, code=killed, status=11/SEGV
Jul  4 18:05:59 raspberrypi systemd[1]: vzlogger.service: Failed with result ‚signal'.
`

The version of the vzlogger is 0.8.0

mbehr1, 2020-07-04:
Hi,

hmm. bad. Can you create a core dump?

Morphi08, 2020-07-04:
I have to change some settings on the raspi and reboot, than I check that there is some core-dumps after the crash; I will submit it

Morphi08, 2020-07-04:
[core-vzlogger-sig11-user0-group0-pid727-time1593885104.gz](https://github.com/volkszaehler/vzlogger/files/4873537/core-vzlogger-sig11-user0-group0-pid727-time1593885104.gz)

mbehr1, 2020-07-05:
Thanks! Can you create the backtrace as well from your build?

gdb bt full

Morphi08, 2020-07-05:
hm, sorry .. I doesn't know this command; it won't get any information with this commands gdb bt full
when I use gdb, in the program bt -> there is no stack ?
can you provide me to use the right command?

mbehr1, 2020-07-05:
Hi,

it should be :
gbd <full path to vzlogger binary> <your core dump file>

then inside gbd
bt 
(or bt full)

Mit freundlichen Grüßen / Best Regards
Matthias Behr

Sent from a mobile device.

Morphi08, 2020-07-05:
Hallo, das sollte jetzt gehen, nachdem man das "set on logging" eingeschaltet hat ;-)
[gdb.txt](https://github.com/volkszaehler/vzlogger/files/4874818/gdb.txt)
Ich hoffe die Datei kann helfen.

Hi, it would be work when the logging is set on into gdb; I hope the textfile can help

Morphi08, 2020-07-05:
i use for a workaround the crontab with this script
`TEST=`ps` -A | grep vzlogger | awk '{ print $4 }'`
if [ "$TEST" != "vzlogger" ]; then
      sleep 5
       systemctl start vzlogger
fi
`

mbehr1, 2020-07-05:
(ich wechsele auch mal auf deutsch :-)

Danke!

Crash ist in der libsml. Schaue ich mal, ob ich da nen Fehler finden kann.
Wenn das bei dir reproduzierbar ist, kann ich eine debug-Version von libsml erstellen, die mehr Log-Ausgaben macht und du erstellst noch mal log/coredump/backtrace?

Gruß
Matthias

 sml_message_parse (buf=0x74d049c8) at src/sml_message.c:71
        msg = 0x74d075f0
        msg_start = 51
        len = 20
#1  0x0012b8cc in sml_file_parse (buffer=0x75edf9c8 "v\vESYA\037\350\005\265\203\021b", buffer_len=372) at src/sml_file.c:54
        file = 0x74d03240
        buf = 0x74d049c8
        msg = 0x74d01fb0
#2  0x000d33d4 in MeterSML::read (this=0x1241418, rds=std::vector of length 32, capacity 32 = {...}, n=32) at /home/pi/vzlogger/src/protocols/MeterSML.cpp:280
        buffer = "\033\033\033\033\001\001\001\001v\vESYA\037\350\005\265\203\021b\000b\000rc\001\001v\001\004ESY\bESY+\261\203\021\v\t\001ESY\021\003\246\037\350\001\001c\035R\000v\vESYA\037\350\005\265\203\022b\000b\000rc\t\000\001ESY\021\003\246\037\350\a\001\000b\n\377\377rb\001e\001\347+\261yw\a\201\201ǂ\003\377\001\001\001\001\004ESY\001w\a\001\000\000\000\t\377\001\001\001\001\v\t\001ESY\021\003\246\037\350\001w\a\001\000\001\b\000\377d\000\000\200\001b\036R\374Y\000\000\000\tr\231\221\004\001w\a\001\000\020\a\000\377\001\001b\033R\376Y\000\000\000\000\000\000i\231\001w\a\001\000$"...
        bytes = 388
        m = 0
        file = 0x75ee1be4
        body = 0x75ee3820
        entry = 0x7eeee6ac <__vdso_gettimeofday+12>
#3  0x000bd7a0 in Meter::read (this=0x1241118, rds=std::vector of length 32, capacity 32 = {...}, n=32) at /home/pi/vzlogger/src/Meter.cpp:246

andig, 2020-07-05:
Lässt sich rausfinden mit welcher Version das compiliert wurde? Ich hätte am ehesten einen der letzten PRs im Verdacht, z.b. diejenigen die mit den CRC Algorithmen rumspielen?

Viele Grüße, Andreas

Morphi08, 2020-07-05:
Hallo andig, wenn Du mir sagst, nach was ich schauen soll, lässt sich das machen ..
@Matthias, durch meinen "Watchdog" läuft die Messung zwar, der vzlogger ist 18x heute abgestürzt..; wenn Du eine Lib entsprechend anpassen kannst, teste ich diese gerne.

VG Arne

mbehr1, 2020-07-11:
Ich kann mit den Backtrace Buffer den Crash reproduzieren. 
Füge grad nen Unit Test hinzu (und dann den Fix :-)

TEST(sml_file, parse) {
	unsigned char buffer =
		"\033\033\033\033\001\001\001\001v\vESYA\037\350\005\265\203\021b\000b\000rc\001\001v\001"
		"\004ESY\bESY+"
		"\261\203\021\v\t\001ESY\021\003\246\037\350\001\001c\035R\000v\vESYA\037\350\005\265\203"
		"\022b\000b\000rc\t\000\001ESY\021\003\246\037\350\a\001\000b\n\377\377rb\001e\001\347+"
		"\261yw\a\201\201ǂ\003\377\001\001\001\001\004ESY\001w\a\001\000\000\000\t\377\001\001\001"
		"\001\v\t\001ESY\021\003\246\037\350\001w\a\001\000\001\b\000\377d\000\000\200\001b\036R"
		"\374Y\000\000\000\tr\231\221\004\001w\a\001\000\020\a\000\377\001\001b\033R\376Y\000\000"
		"\000\000\000\000i\231\001w\a\001\000$";
	size_t bytes = 388;
	// bytes and buffer like from sml_transport_read
	sml_file *file = sml_file_parse(buffer + 8, bytes - 16);
	TEST_ASSERT_NOT_NULL(file);
	TEST_ASSERT_EQUAL(1, file->messages_len);
	sml_file_free(file);
}

Ist es ok, wenn ich da deine Daten aufnehme? (also den Bufferinhalt)

mbehr1, 2020-07-11:
FIx ist hier:
https://github.com/volkszaehler/libsml/pull/85 <https://github.com/volkszaehler/libsml/pull/85>

Kannst du den ausprobieren? (compile libsml from that pr, install and rerun vzlogger and see whether you get a crash?)

Btw: Wäre interessant zu sehen, ob du auch eine Meldung (auf der console, d.h. müsstest vzlogger in non-daemon mode starten) bekommst:
error: message type 0900 not yet implemented…

Ich bin mir aber nicht sicher, ob das nicht ein Testfehler ist, weil die Buffer Daten unten nicht komplett sind (leider keine 388 bytes, weniger, war zu faul zum Zählen…). Wenn du mir mal nen vollen Dump schicken könntest, könnte ich den Unit Test anpassen und das Parsen der Daten gleich implementieren…).

Gruß
Matthias

devZer0, 2020-07-11:
können wir einen binary-dump von deinem zähler haben? für https://github.com/devZer0/libsml-testing ?

Morphi08, 2020-07-11:
Hallo Matthias,
Ich glaube ich bin nicht firm genug im Gebrauch von dem Github? ;-)
Ich habe in der Annahme es richtig zu machen nun folgende Befehle genutzt:

Cd vzlogger/libs/libsml
Git pull
Cd ../..
./install.sh libsml

Und danach den vzlogger nicht mehr als Damon gestartet; es kommt keine Fehlermeldung 0900
VG
Arne

mbehr1, 2020-07-11:
3 weitere (potentielle) Fixes sind hier: https://github.com/volkszaehler/libsml/pull/86/files

mbehr1, 2020-07-11:
@Morphi08 
Hallo Arne,
kam beim install.sh die Ausgabe:
building and installing libsml
?
Kannst du mal in 
cd vzlogger/libs/libsml
und dort
git pull
git log
make && make install
aufrufen?

Das nimmt zwar nicht den PR 86 mit (aber der erste ist bereits gemerged).

Tritt dein Fehler noch auf?

Morphi08, 2020-07-11:
Hallo Matthias,

Hier die ersten Logeinträge:
sudo git log
commit f26d7d89aca7a65b8cfd1607ac19f262423cbabb (HEAD -> master, origin/master, origin/HEAD)
Merge: 6609c81 406b331
Author: Matthias Behr <mbehr@mcbehr.de>
Date:   Sat Jul 11 16:35:09 2020 +0200

    Merge pull request #85 from volkszaehler/fix_crash_vzlogger_crc
    
    Fix crash accessing null crc

commit 406b331a7557571ee458cebed042d01b1edc8a2d (origin/fix_crash_vzlogger_crc)
Author: Matthias Behr <mbehr@mcbehr.de>
Date:   Sat Jul 11 12:57:56 2020 +0200

    Fix crash accessing null crcsml_u16_parse can return null.So msg->crc can be null and needs to be verified before comparision.Issue: vzlogger_crash_report

commit 6609c8117ba2c987aea386a7fffb9b4746636be6
Author: Michael Rüttgers <github@michael-ruettgers.de>
Date:   Thu Apr 9 08:45:11 2020 +0200

Und hier das make && make install:
make[1]: Verzeichnis „/home/pi/vzlogger/libs/libsml/test“ wird verlassen
make[1]: Verzeichnis „/home/pi/vzlogger/libs/libsml/sml“ wird betreten
make[1]: Für das Ziel „all“ ist nichts zu tun.
make[1]: Verzeichnis „/home/pi/vzlogger/libs/libsml/sml“ wird verlassen
make[1]: Verzeichnis „/home/pi/vzlogger/libs/libsml/examples“ wird betreten
make[1]: Für das Ziel „all“ ist nichts zu tun.
make[1]: Verzeichnis „/home/pi/vzlogger/libs/libsml/examples“ wird verlassen
make[1]: Verzeichnis „/home/pi/vzlogger/libs/libsml/test“ wird betreten
make[2]: Verzeichnis „/home/pi/vzlogger/libs/libsml/sml“ wird betreten
make[2]: Für das Ziel „all“ ist nichts zu tun.
make[2]: Verzeichnis „/home/pi/vzlogger/libs/libsml/sml“ wird verlassen
Unity test run 1 of 1
libsml: error: message type 0900 not yet implemented
libsml: NYI: sml_message_body_free for message type 0900
libsml: warning: could not read the whole file
........................................................................................
-----------------------
88 Tests 0 Failures 0 Ignored
OK
make[1]: Verzeichnis „/home/pi/vzlogger/libs/libsml/test“ wird verlassen
make[1]: Verzeichnis „/home/pi/vzlogger/libs/libsml/sml“ wird betreten
install -d /usr/local/lib
install: die Zugriffsrechte von „/usr/local/lib“ können nicht geändert werden: Die Operation ist nicht erlaubt
make[1]: *** [Makefile:86: install] Fehler 1
make[1]: Verzeichnis „/home/pi/vzlogger/libs/libsml/sml“ wird verlassen
make: *** [Makefile:14: install] Fehler 2

Mit dem Sudo geht dann anschliessend auch das make install; ich probiere nun den vzlogger mal aus, wie lange der durchhält..

VG

Morphi08, 2020-07-11:
Ist eben wieder abgestürzt:
i@raspberrypi:/var/lib/coredumps $ sudo gdb /var/local/vzlogger core-vzlogger-sig11-user0-group0-pid3042-time1594491503 
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
/var/local/vzlogger: Datei oder Verzeichnis nicht gefunden.
[New LWP 3046]
[New LWP 3048]
[New LWP 3049]
[New LWP 3042]
[New LWP 3050]
[New LWP 3051]
[New LWP 3047]
Core was generated by `/usr/local/bin/vzlogger -c /etc/vzlogger.conf'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0012cbf0 in ?? ()
[Current thread is 1 (LWP 3046)]
(gdb) bt full
#0  0x0012cbf0 in ?? ()
No symbol table info available.
#1  0x0012cbdc in ?? ()
No symbol table info available.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

mbehr1, 2020-07-11:
Hmm. Und Backtrace nicht aussagekräftig.

Kannst du dann doch mal den PR 86 ausprobieren:

in vzlogger/lib/libsml wechseln

da

git fetch origin
git checkout -b libFuzzer_fixes origin/libFuzzer_fixes

dann 
make && make install

dann vzlogger neu bauen (bin mir nicht sicher, ob wir statisch oder dynamisch linken…)

cd vzlogger 
make clean && make && sudo make install

dann nochmal versuchen und ggf. weiteren coredump mit besseren backtrace…

(Achtung: danach wieder libsml auf normalen master umstellen:
cd vzlogger/lib/libsml
git checkout master
git pull
make && make install

Danke!

mbehr1, 2020-07-11:
@Morphi08  ich glaube, du musst vzlogger auch neu compilieren:

#0 sml_message_parse (buf=0x755059c8) at src/sml_message.c:71
warning: Source file is more recent than executable.

oder min. ein libsml make && make install

Morphi08, 2020-07-12:
Hallo Matthias,

Ich habe heute nach nach Deiner Anleitung das PR86 eingebunden und dann auch den vzlogger neu compiliert, jedoch noch nicht deine „Achtung:“ auf Master umgestellt, da ich dachte, das dann wieder automatisch diese Libsml genommen wird. Seit 0:21 bis läuft er durch :-)
Mal sehen was die nächste Zeit so passiert 

VG
Arne

mbehr1, 2020-07-12:
Hallo Arne,

prima. Dann scheinen die Fixes zu greifen. Ich finde grad durch Anwendung von libFuzzer noch massenhaft weitere mögliche... :-) Crashes.
Master umstellen nur nicht vergessen, sonst bleibt beim nächsten Update die libsml auf dem Stand.

Ich würde vorschlagen, wir machen das Issue hier zu, sobald PR 86 gemerged ist?

Gruß
Matthias

mbehr1, 2020-07-18:
Fixed with merge of libsml PR #86.

(@Morphi08 jetzt nicht vergessen, wieder auf libsml/master umzustellen)

closed 2020-07-18 by mbehr1