volkszaehler/vzlogger #430: SSL_connect: SSL_ERROR_SYSCALL

milefork, 2020-08-13:
Hi,

I've compiled version c5bcd3296a4400ddf041db21c67a17f4e526f71e of vzlogger on the new Raspi OS.

Everything works fine except the push to my middleware which is secured by https.

The error I'm getting is:

[Aug 13 11:19:00][chn1] CURL: Connected to XXXXX.info (XXXXX) port 443 (#1)
[Aug 13 11:19:00][chn1] CURL: ALPN, offering h2
[Aug 13 11:19:00][chn1] CURL: ALPN, offering http/1.1
[Aug 13 11:19:00][chn1] CURL: successfully set certificate verify locations:
[Aug 13 11:19:00][chn1] CURL:   CAfile: /etc/ssl/certs/ca-certificates.crt
[Aug 13 11:19:00][chn1] CURL: Sent 5 bytes..
[Aug 13 11:19:00][chn1] CURL: Sent '' bytes
[Aug 13 11:19:00][chn1] CURL: TLSv1.2 (OUT), TLS handshake, Client hello (1):
[Aug 13 11:19:00][chn1] CURL: Sent 224 bytes..
[Aug 13 11:19:00][chn1] CURL: Sent '' bytes
[Aug 13 11:19:00][chn1] CURL: OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to XXXX:443
[Aug 13 11:19:00][chn1] CURL: Closing connection 1

As you might see I've actually done some tests, because my server can only handle TLS1.2. But it seems that is has nothing to do with the TLS version. Also the vzclient works like a charm.

The changes I've tested with were:

https://gist.github.com/milefork/6a286e9cda1fb4a1461d62563b154867

Any Ideas?

milefork, 2020-08-28:
Ums mal eben in deutsch zu verfassen:
Ich habe das den vzlogger noch mal auf nem normalen x64 gebuildet und mir das ganze da nochmal angeschaut. Da geht es auch kaputt.
Eine Verbindung mit dem openssl s_client funktiontiert ohne Probleme.

Der log vom vzlogger: https://gist.github.com/milefork/dca460081f8dbb415d0b0b61a7e1f199
Ich hab das mal in wireshark mir angeschaut:
[image]
Wie man in der vorletzen Zeile sieht, schickt der Client auf einmal einen Hello Request, welcher lauf RFC gar nicht vom Client aus gesendet werden darf(https://tools.ietf.org/html/rfc5246#section-7.4.1.1)

Jemand ne Idee? Ich weiß der logger nutzt einfach curl, aber kann es sein das da irgend eine "schlechte" version o.ä verwendet wird? Denn mit wget, curl und openssl komme ich ohne Probleme drauf.

milefork, 2020-08-31:
Noch ein Update:
Habe das ganze mal gegen https://demo.volkszaehler.org/middleware.php laufen lassen. Das gleiche Problem.
Im oberen Teil ist der vzlogger, Im unteren die Verbindung wenn man im Terminal:
`curl https://demo.volkszaehler.org/middleware.php` laufen lässt.
Also mit dem curl befehl geht es, mit dem vzlogger nicht.
[image]

Mache ich jetzt ein Issue bei curl auf oder liegt es am logger?

andig, 2020-08-31:
Fehlen da irgendwie aktuelle ca root Zertifikate?

mbehr1, 2020-08-31:
ich schaue mir das mal an.

mbehr1, 2020-09-17:
kann es sein, dass es das gleiche wie hier ist: https://github.com/openssl/openssl/issues/9566

milefork, 2020-09-17:
Also ich habe das mal mit einer Main getestet, welche eure curl Zeilen benutzt. Da funktioniert das ohne Probleme. Ich glaube das Problem liegt vielleicht eher an eurem Multithreading?

https://gist.github.com/milefork/79e33cae628ca5fdd4af9a799f852ced

hupling, 2021-02-05:
```
[Feb 05 11:07:22][chn0] CURL: Hostname example.com was found in DNS cache
[Feb 05 11:07:22][chn0] CURL:   Trying xx.xx.x.xx...
[Feb 05 11:07:22][chn0] CURL: TCP_NODELAY set
[Feb 05 11:07:22][chn0] CURL: Connected to example.com (xx.xx.xx.xx) port 443 (#48)
[Feb 05 11:07:22][chn0] CURL: ALPN, offering http/1.1
[Feb 05 11:07:22][chn0] CURL: successfully set certificate verify locations:
[Feb 05 11:07:22][chn0] CURL:   CAfile: /etc/ssl/certs/ca-certificates.crt
[Feb 05 11:07:22][chn0] CURL: Sent 5 bytes.. 
[Feb 05 11:07:22][chn0] CURL: Sent '' bytes
[Feb 05 11:07:22][chn0] CURL: TLSv1.3 (OUT), TLS handshake, Client hello (1):
[Feb 05 11:07:22][chn0] CURL: Sent 512 bytes.. 
[Feb 05 11:07:22][chn0] CURL: Sent '' bytes
[Feb 05 11:07:22][chn0] CURL: Received 5 bytes
[Feb 05 11:07:22][chn0] CURL: Received '' bytes
[Feb 05 11:07:22][chn0] CURL: TLSv1.3 (IN), TLS alert, Server hello (2):
[Feb 05 11:07:22][chn0] CURL: Received 2 bytes
[Feb 05 11:07:22][chn0] CURL: Received '
' bytes
[Feb 05 11:07:22][chn0] CURL: error:140943F2:SSL routines:ssl3_read_bytes:sslv3 alert unexpected message
[Feb 05 11:07:22][chn0] CURL: stopped the pause stream!
[Feb 05 11:07:22][chn0] CURL: Closing connection 48
[Feb 05 11:07:22][chn0] CURL: SSL connect error
[Feb 05 11:07:23][mtr0] Got 1 new readings from meter:
[Feb 05 11:07:23][mtr0] Reading: id=NilIdentifier/NilIdentifier value=20.62 ts=1612519643671
[Feb 05 11:07:23][chn0] Adding reading to queue (value=20.62 ts=1612519643671)
```
Ich habe irgendwie so ein ähnliches Problem mit TLS 1.3

PR #446 (2021-02-07): SSL Connect Error Volkszähler API
closes #430 
Das Pushen auf die Volkszähler API Middleware.php mit HTTPS hat nicht funktioniert. Das Auskommentieren einer Zeile behebt das Problem.

```
[Feb 05 11:07:22][chn0] CURL: Hostname example.com was found in DNS cache
[Feb 05 11:07:22][chn0] CURL:   Trying xx.xx.x.xx...
[Feb 05 11:07:22][chn0] CURL: TCP_NODELAY set
[Feb 05 11:07:22][chn0] CURL: Connected to example.com (xx.xx.xx.xx) port 443 (#48)
[Feb 05 11:07:22][chn0] CURL: ALPN, offering http/1.1
[Feb 05 11:07:22][chn0] CURL: successfully set certificate verify locations:
[Feb 05 11:07:22][chn0]  [... cut]
closed 2021-02-07 by mbehr1