volkszaehler/vzlogger #433: Parameter interval is ignored in protocol exec

bbr111, 2020-09-09:
It seems that the interval parameter is ignored in protocol exec:

In the logfile you can see multiple reading in the same second. Interval is set to 10 in config file.

logfile:
[Sep 09 21:32:47][exec] MeterExec::read: Calling '/home/pi/get_pv_power.sh'
[Sep 09 21:32:47][exec] MeterExec::read: Reading line: 'PV = 0'
[Sep 09 21:32:47][exec] MeterExec::read: string: PV, value: 0.000000, timestamp: -1.000000
[Sep 09 21:32:47][exec] MeterExec::read: Closing process '/home/pi/get_pv_power.sh'
[Sep 09 21:32:47][mtr1] Got 1 new readings from meter:
[Sep 09 21:32:47][mtr1] Reading: id=PV/StringIdentifier: value=0.00 ts=1599679967197
[Sep 09 21:32:47][chn1] Adding reading to queue (value=0.00 ts=1599679967197)
[Sep 09 21:32:47][exec] MeterExec::read: Calling '/home/pi/get_pv_power.sh'
[Sep 09 21:32:47][exec] MeterExec::read: Reading line: 'PV = 0'
[Sep 09 21:32:47][exec] MeterExec::read: string: PV, value: 0.000000, timestamp: -1.000000
[Sep 09 21:32:47][exec] MeterExec::read: Closing process '/home/pi/get_pv_power.sh'
[Sep 09 21:32:47][mtr1] Got 1 new readings from meter:
[Sep 09 21:32:47][mtr1] Reading: id=PV/StringIdentifier: value=0.00 ts=1599679967341

config:

        {
            "enabled": true,               // disabled meters will be ignored (default)
            "skip": true,                  // errors when opening meter may be ignored if enabled
            "interval": 10,
            "protocol": "exec",              // meter protocol, see 'vzlogger -h' for full list
            "command": "/home/pi/get_pv_power.sh",
            "format": "$i = $v",
            "aggtime": 60,                  // aggregate meter readings and send middleware update after <aggtime> seconds

            "channels": [{
                "uuid": "6edfd960-f2c4-11ea-88a8-3df347774982",
                "api": "Volkszaehler",
                "middleware": "http://localhost/middleware.php",
                //"aggmode": "none",
                "identifier": "PV",
                "aggmode": "AVG",
            }] //channel end
        }

frankrichter, 2020-09-10:
Please try without aggtime. If i remember correctly the aggtime loop ist
inside the interval loop.

bbr111, 2020-09-10:
Without aggtime it works. 

My intention was to read the Logvalues from PV every 10 seconds, calculate an average an write this value to database every 60 seconds. This is to minimize the amount of data. The system is running on raspi with sdcard only. to many write operations will kill the sdcard.

also i try the MeterFile with interval: -1 and agetime 60.
- create an tmpfs with 5 MB
- create 6 cronjobs with sleep 10 && cmd, sleep 20 & cmd ,... >outputfile_to_tmpfs
- create meter file with interval -1 , aggtime 60, path outputfile

Every time the file changes, the channel gets a new reading. ( 10 seconds interval from cron )
6 readings will be averaged and write to database.

But it would be nice, if interval and aggtime can be used the same time in Meter exec.

frankrichter, 2020-09-10:
You could add a sleep 10 to your script as a workaround.

bbr111, 2020-09-10:
Good idea, sometimes I think too complicated,
Would you like have this issue open?

KanteTaete, 2020-12-30:
I face the same problem.

If parameter `aggtime` is set to `-1` for the meter, then the parameter `interval` (in my case 30s) is regarded, if `aggtime` is set to `60`, then the command is executed continuously (next execution directly after previous finished).

Interesting behaviour:
After writing data to the InfluxDB (at 22:37:08), the 30s delay is kept (next execution at 22:37:38, then again immediately after reading and adding to queue is finished).

So it seems that the `interval` is implemented "at the wrong place"...

Here parts of my logfile:
```
[Dec 30 22:36:56][exec] MeterExec::read: Calling '<pythonscript>'
...
[Dec 30 22:37:00][exec] MeterExec::read: Closing process '<pythonscript>'
...
[Dec 30 22:37:00][chn2] Adding reading to queue (value=2300.00 ts=1609364218274)
...
[Dec 30 22:37:00][exec] MeterExec::read: Calling '<pythonscript>'
...
[Dec 30 22:37:08][exec] MeterExec::read: Closing process '<pythonscript>'
...
[Dec 30 22:37:08][chn2] Adding reading to queue (value=2300.00 ts=1609364225703)
...
[Dec 30 22:37:08][AVG]  [0] 2298.000000 @ 1609364169861
[Dec 30 22:37:08][AVG]  [1] 2299.000000 @ 1609364173621
[Dec 30 22:37:08][AVG]  [2] 2299.000000 @ 1609364177342
[Dec 30 22:37:08][AVG]  [3] 2299.000000 @ 1609364181047
[Dec 30 22:37:08][AVG]  [4] 2297.000000 @ 1609364184771
[Dec 30 22:37:08][AVG]  [5] 2298.000000 @ 1609364188479
[Dec 30 22:37:08][AVG]  [6] 2299.000000 @ 1609364192202
[Dec 30 22:37:08][AVG]  [7] 2298.000000 @ 1609364195913
[Dec 30 22:37:08][AVG]  [8] 2299.000000 @ 1609364199623
[Dec 30 22:37:08][AVG]  [9] 2301.000000 @ 1609364203340
[Dec 30 22:37:08][AVG]  [10] 2299.000000 @ 1609364207057
[Dec 30 22:37:08][AVG]  [11] 2301.000000 @ 1609364210764
[Dec 30 22:37:08][AVG]  [12] 2300.000000 @ 1609364214520
[Dec 30 22:37:08][AVG]  [13] 2300.000000 @ 1609364218274
[Dec 30 22:37:08][AVG]  [14] 2300.000000 @ 1609364225703
[Dec 30 22:37:08][AVG]  [15] RESULT 2299.134666 @ 1609364225703
[Dec 30 22:37:08][chn2] Buffer has 1 items
...
[Dec 30 22:37:08][mtr0] Next reading in 30 seconds
...
[Dec 30 22:37:08][chn2] InfluxDB CURL success
...
[Dec 30 22:37:38][exec] MeterExec::read: Calling 'pythonscript'
...
[Dec 30 22:37:42][exec] MeterExec::read: Closing process 'pythonscript'
...
[Dec 30 22:37:42][chn2] Adding reading to queue (value=2301.00 ts=1609364259948)
...
[Dec 30 22:37:42][exec] MeterExec::read: Calling 'pythonscript'
```

r00t-, 2021-03-09:
(note that `interval` in itself is rather primitive... especially with `protocol=exec`, if the execution time is non-zero, it is not substracted from the interval, leading to a slowly shifting readings period. - maybe open a separate ticket or MR for that.)

bbr111, 2021-11-01:
If tested the fix, and it works good.

[Nov 01 12:10:09][mtr1] waiting 10 seconds before next reading

for my config it should make a reading every 10 second and aggtime is set to 60
This is working for me.

Thank you!

PR #454 (2021-11-05): fix combined use of `interval` and aggregation
it seems combined use of `interval` and aggregation was never considered.
the expected use-case of aggregation is to reduce the (externally determined)
rate at which a meter is generating data.
but some users want to use it create averages over data collected at a
configured `interval` - so let's support that.

fix #433
closed 2021-11-05 by r00t-