volkszaehler/vzlogger #436: Timestamp not working

kickbutts, 2020-11-14:
Hi there,
i installed vzlogger but i got some problems concerning the timestamp.
The log of vzlogger is shwoing this:
`[Nov 14 19:35:40][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=390069.10 ts=0
[Nov 14 19:35:40][mtr0] Reading: id=1-0:2.8.0*255/ObisIdentifier:1-0:2.8.0*255 value=205246.90 ts=0
[Nov 14 19:35:40][mtr0] Reading: id=1-0:16.7.0*255/ObisIdentifier:1-0:16.7.0*255 value=700.30 ts=0
[Nov 14 19:35:40][chn0] Adding reading to queue (value=700.30 ts=0)
`
The timestamp are zero (ts=0). So when sensing my Data to influxdb they wont be stored because of a missing timestamp.
Any ideas?
Thanks!

J-A-U, 2020-11-14:
> The timestamp are zero (ts=0). So when sensing my Data to influxdb
> they wont be stored because of a missing timestamp.
> Any ideas?

"use_local_time"=true;
https://wiki.volkszaehler.org/software/controller/vzlogger/vzlogger_conf_parameter#sml

kickbutts, 2020-11-15:
That was the point. Thanks! It is working now.
Maybe you could add it as an option to the vzlogger Config Editor Page. This might help.

closed 2020-11-15 by kickbutts

J-A-U, 2020-11-15:
> Maybe you could add it as an option to the vzlogger Config Editor Page.

It's already implemented.