volkszaehler/vzlogger #439: Influxdb v2 api and Token instead of basic auth

Waldmeisda, 2020-12-13:
Hi,

is it possible to support the newer API? (api/v2/write?org?xxxx&bucket=vzlogger instead of plain /write

and instead of username + password in basic auth, just use the 

Authorization: Token <token hash>

/src/api/InfluxDB.cpp:

Add the header as follows in https://curl.se/libcurl/c/CURLOPT_HTTPHEADER.html

and modify the entry point as described in https://docs.influxdata.com/influxdb/v2.0/write-data/developer-tools/api/

I've hardcoded these settings in a local copy of InfluxDB.cpp seams to work with influxdb2

BR

Frank

astrakid, 2021-02-28:
not tested yet, but it seems that vzlogger already supports token authentication:

"meters": [
        // examples for InfluxDB as storage
        {
            // See vzlogger.conf for complete meter configuration options

            "enabled": true,                 // disabled meters will be ignored
            "protocol": "sml",               // see 'vzlogger -h' for list of available protocols
            "device": "/dev/ttyAMA0",
            "channels": [{
                "api": "influxdb", // use the InfluxDB api
                "uuid": "01234567-9abc-def0-1234-56789abcdefe", // use the uuid command to generate this
                "identifier" : "1-0:16.7.0", // OBIS code for "power" 
                "host": "127.0.0.1:8086",                       // This assumes that InfluxDB is running on localhost
                //"database": "vzlogger",                       // Optional: make sure this database exists in InfluxDB
                //"measurement_name": "vz_measurement",         // Optional: It is recommended that all your meters have the same InfluxDB measurement name
                //"tags": "foo=bar,example=42",                 // Optional: Additional tags to append when inserting data
                //"username": "example",                        // Optional: When InfluxDB Auth is enabled you need to set the correct user and password
                //"password": "secure",
                //"token": "login token",                       // Optional: Use login token instead of username. Remove sername from config to use token. 
                //"organization": "example.com",                // Optional: Configure organization to use /api/v2/write?org=xxx&bucket=database entry point
                //"max_batch_inserts": 4500,                    // Optional: Max number of measurements per request. No need to change this
                //"max_buffer_size": 450000,                    // Optional: Max number of measurements to be cached when InfluxDB is not available
                //"timeout": 30,                                // Optional: Time in seconds after which requests to InfluxDB time out
                //"send_uuid": false,                           // Optional: Disables the sending of the UUID to the InfluxDB server
                //"ssl_verifypeer": false,                      // Optional: Disables the certificate verification for https connections
            }]
        },
    ]

Waldmeisda, 2021-02-28:
Yes, because I added it. https://github.com/volkszaehler/vzlogger/pull/440
Can be closed btw:

Frank

closed 2021-02-28 by Waldmeisda