volkszaehler/vzlogger #442: log file per day (or week)

isarrider, 2021-01-26:
Hallo, 

I have a smart meter, that is unreliable...
Is there a way to generate a new log file per day
(in my case, the I can delete it after I checked this day)

Thx!

KanteTaete, 2021-01-26:
Hallo Isarrider!

If you use Linux or Raspi-OS: Yes. 
You can use the Linux built-in mechanism logrotate which is used for e.g. the syslog file as well.
You can find details searching for "logrotate" in the net, e.g. here: [ubuntuusers.de](https://wiki.ubuntuusers.de/Logdateien/#Logrotate) (German)

My file looks like this:
```
/var/log/vzlogger.log {
    daily
    rotate 7
    missingok
    dateext
    copytruncate
    compress
}
```
For detailed options, see the linux manuals in the internet.

Regards!
Kante Täte

isarrider, 2021-01-27:
I will try it...
lets see if copytruncate does the job well ;)
Thank you!

closed 2021-01-27 by J-A-U

isarrider, 2021-01-28:
I tried it. it works - thankyou...
(still ugly with the copytruncate)

r00t-, 2021-03-14:
@isarrider:
you might want to try https://github.com/volkszaehler/vzlogger/pull/458

isarrider, 2021-03-14:
awesome - thx!