volkszaehler/vzlogger #443: Suggestion for additional API type

martink173, 2021-01-27:
I am using a two-way meter, and the reading is either positive or negative, which is fine. However, for some other reasons, I want to have two separate channels, one for energy consumption (positive), and one for energy supply (negative values, inverted). Therefore it would be nice to feed the data coming from IR to e.g. a PHP script, and the PHP script can send it back to vzlogger middleware. This would result in the possibility for doing any type of calculations with read values.

A new PHP-API could do this trick? I tried to set "api" in vzlogger.conf to "volkszaehler" and point it to my own PHP script, but this doesn't work, because vzlogger expects some kind of answer from the script, which I didn't emulate.

J-A-U, 2021-01-27:
Feel free to implement a solution for this special use case and provide a pull request.

If you are not able to do so i suggest a closer look on NodeRed:
https://wiki.volkszaehler.org/howto/node-red

martink173, 2021-01-27:
I was not aware of node-red, thanks for that.

Maybe someone can also point me in the direction, which answer the middleware sends back, when receiving json data from a channel.

J-A-U, 2021-01-27:
Take a look in the vzlogger.log with verbosity set to 15. The JSON send back from Middleware is easy to find.
Or simply try by yourself: https://wiki.volkszaehler.org/development/api/reference

closed 2021-01-28 by J-A-U

martink173, 2021-01-28:
OK, got it now. Thanks for pointing that out. For future reference, one may imitate middleware behavior by sending a JSON structure containing an array of the version and affected rows from a PHP script. That script processes measurement values, does some calculations and sends it back to middleware.