volkszaehler/vzlogger #444: "dump_file" not working anymore

isarrider, 2021-01-29:
Hi,

"dump_file" is for example documented here:
https://wiki.volkszaehler.org/howto/edl-ehz_unknown

When dumping it in here, it reports invalid...
https://volkszaehler.github.io/vzlogger/

I would like to be able to use it
(with all meters, cause I have d0 and smal)...

Thoughts?
Thx!

isarrider, 2021-01-29:
nevermind the not working, typo... ;)

But still, the dump_raw is interesting for SML...

J-A-U, 2021-01-29:
Dump never worked with sml, it's only implemented for d0 meter type.
And you don't need it after communication is established.

isarrider, 2021-01-29:
I do, as the Meter is unreliable and I need the raw data...
Why did it neber work btw?

J-A-U, 2021-01-29:
Nobody implemented it, easy as that.

For unreliable reading check the position of the reading head.

isarrider, 2021-02-08:
It is not unreliable, the meter is with beta firmware...
but I circumvented it now with loglevel=5

closed 2021-02-08 by isarrider