volkszaehler/vzlogger #445: Implement scaler for all sources

RalfGebhard, 2021-02-02:
Scaler currently doesn’t work for sml meters,
But for scaling meters “Messwandelzähler”, their output is a fragment of the real output.

Ralf

r00t-, 2021-03-09:
so you are refering to the `scaler` channel property,
which is only implemented for MeterOCR
https://github.com/volkszaehler/vzlogger/blob/1a6e10b6d8d69282f3df8152c35ce1b1f19baad7/etc/vzlogger_generic.schema.json#L743
https://github.com/volkszaehler/vzlogger/blob/1a6e10b6d8d69282f3df8152c35ce1b1f19baad7/src/protocols/MeterOCR.cpp#L132
and the mysmartgrid API:
https://github.com/volkszaehler/vzlogger/blob/1a6e10b6d8d69282f3df8152c35ce1b1f19baad7/etc/vzlogger_generic.schema.json#L151
https://github.com/volkszaehler/vzlogger/blob/1a6e10b6d8d69282f3df8152c35ce1b1f19baad7/src/api/MySmartGrid.cpp#L84

actually this looks like a BUG to me:
if one was logging an OCR-meter to a mysmartgrid server, the scaler would be applied twice?!

the implementation should probably be moved to a central place (the `Reading{}` class?),
where it would apply to all meters, instead of only specific meters or apis.

r00t-, 2021-03-11:
there seems to be a partial fix in https://github.com/volkszaehler/vzlogger/issues/245