volkszaehler/vzlogger #449: vzlogger keeps running in main loop

juergen4mueller, 2021-02-21:
I use vzlogger version 0.8.0.
 based on git version: heads/master-0-g033de9dc7d
 last commit date: Wed, 10 Feb 2021 14:44:17 +0100

If I start vzlogger manually it runs, but stays still in main loop. I have to end the run by strg-c and get the message:
Here the output:
[Feb 21 19:21:30][chn2] emptied all (1) values
^C[Feb 21 19:21:39][]     main loop indicating all mappings to quit
[Feb 21 19:21:39][mtr0] MeterMap::cancel entered...
[Feb 21 19:21:39][mtr0] MeterMap::cancel wait for readingthread
[Feb 21 19:21:39][mtr0] MeterMap::cancel wait for meter::close
[Feb 21 19:21:39][mtr0] MeterMap::cancel finished.
[Feb 21 19:21:39][]     Server stopped.
[Feb 21 19:21:39][]     Trying to delete curlSessionProvider...
[Feb 21 19:21:39][]     deleted curlSessionProvider

I want to run it with cron to be in sync with other loggers. Therefore it is important to fix my problem.
For now I only read in 4 one wire sensors, in short future I will add the SML-meters.

My config is here:
{
  "retry": 0,
  "daemon": false,
  "verbosity": 15,
  "log": "/var/log/vzlogger.log",
  "push": [],
  "local": {
    "enabled": false,
    "port": 8080,
    "index": false,
    "timeout": 0,
    "buffer": 0
  },
  "meters": [
    {
      "enabled": true,
      "allowskip": false,
      "channels": [
        {
          "api": "volkszaehler",
          "uuid": "d9e015e0-733c-11eb-a7fd-c3e2efe5b3df",
          "identifier": "28-000004586ff3",
          "middleware": "http://localhost/middleware.php",
        },
        {
          "api": "volkszaehler",
          "uuid": "7a1c1980-733d-11eb-a15c-67a8d56b6336",
          "identifier": "28-0000045889aa",
          "middleware": "http://localhost/middleware.php",
        },
        {
          "api": "volkszaehler",
          "uuid": "a09130c0-733d-11eb-a205-351e39202bfd",
          "identifier": "28-0000045881c4",
          "middleware": "http://localhost/middleware.php",
        },
        {
          "api": "volkszaehler",
          "uuid": "bda83530-733d-11eb-a781-59cd8027de77",
          "identifier": "28-00000459afa8",
          "middleware": "http://localhost/middleware.php",
        }
      ],
      "protocol": "w1therm"
    }
  ]
}

J-A-U, 2021-02-21:
Vzlogger is designed to use as background service. `"daemon": false,` is only suited for pre-run-tests.

r00t-, 2021-02-21:
this is a duplicate of https://github.com/volkszaehler/vzlogger/issues/389

closed 2021-02-21 by andig