volkszaehler/vzlogger #451: can't get information out of iskra MT631

mr-gosh, 2021-02-24:
I think this is the wrong place but i searched for 2 weeks now and i am very frustrated.

I have an ISKRA ISU-MAT MT631-D1A52
I activated the Details with the pin code from the stadtwerke hannover and deactivated the pin completely - so everything looks fine here.
I attached the [ELV USB-IEC Reader](https://de.elv.com/elv-bausatz-lesekopf-mit-usb-schnittstelle-fuer-digitale-zaehler-usb-iec-155523) that is detected via `/sbin/udevadm info --query=all --name=/dev/ttyUSB0` with following Details:

```
P: /devices/platform/soc/20980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/ttyUSB0/tty/ttyUSB0
N: ttyUSB0
L: 0
S: serial/by-path/platform-20980000.usb-usb-0:1.2:1.0-port0
S: serial/by-id/usb-Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_2c3758324ea0ea11a52c5bcbfe9faf31-if00-port0
E: DEVPATH=/devices/platform/soc/20980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/ttyUSB0/tty/ttyUSB0
E: DEVNAME=/dev/ttyUSB0
E: MAJOR=188
E: MINOR=0
E: SUBSYSTEM=tty
E: USEC_INITIALIZED=151427070645
E: ID_VENDOR=Silicon_Labs
E: ID_VENDOR_ENC=Silicon\x20Labs
E: ID_VENDOR_ID=10c4
E: ID_MODEL=CP2102N_USB_to_UART_Bridge_Controller
E: ID_MODEL_ENC=CP2102N\x20USB\x20to\x20UART\x20Bridge\x20Controller
E: ID_MODEL_ID=ea60
E: ID_REVISION=0100
E: ID_SERIAL=Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_2c3758324ea0ea11a52c5bcbfe9faf31
E: ID_SERIAL_SHORT=2c3758324ea0ea11a52c5bcbfe9faf31
E: ID_TYPE=generic
E: ID_BUS=usb
E: ID_USB_INTERFACES=:ff0000:
E: ID_USB_INTERFACE_NUM=00
E: ID_USB_DRIVER=cp210x
E: ID_VENDOR_FROM_DATABASE=Cygnal Integrated Products, Inc.
E: ID_MODEL_FROM_DATABASE=CP2102/CP2109 UART Bridge Controller [CP210x family]
E: ID_PATH=platform-20980000.usb-usb-0:1.2:1.0
E: ID_PATH_TAG=platform-20980000_usb-usb-0_1_2_1_0
E: DEVLINKS=/dev/serial/by-path/platform-20980000.usb-usb-0:1.2:1.0-port0 /dev/serial/by-id/usb-Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_2c3758324ea0ea11a52c5bcbfe9faf31-if00-port0
E: TAGS=:systemd:
```

And my /etc/vzlogger.conf looks like this (but i replaced parts of the uuid by "replaced-by-gibberish":

```
{
  "retry": 30,
  "daemon": true,
  "verbosity": 15,
  "log": "/var/log/vzlogger.log",
  "local": {
    "enabled": false,
    "port": 8081,
    "index": true,
    "timeout": 0,
    "buffer": 0
  },
  "meters": [
    {
      "enabled": true,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
          "api": "volkszaehler",
          "uuid": "55688eb0-replaced-by-gibberish",
          "identifier": "1-0:1.8.0",
          "middleware": "http://10.1.1.11/middleware.php",
          "aggmode": "none",
          "duplicates": 0
        },
        {
          "api": "volkszaehler",
          "uuid": "6c73e920-replaced-by-gibberish",
          "identifier": "1-0:2.8.0",
          "middleware": "http://10.1.1.11/middleware.php",
          "aggmode": "none",
          "duplicates": 0
        },
        {
          "api": "volkszaehler",
          "uuid": "f5ffe6e0-replaced-by-gibberish",
          "identifier": "1-0:16.7.0",
          "middleware": "http://10.1.1.11/middleware.php",
          "aggmode": "none",
          "duplicates": 0
        }
      ],
      "protocol": "sml",
      "device": "/dev/ttyUSB0",
      "baudrate": 9600,
      "parity": "8n1",
      "read_timeout": 60
    }
  ]
}
```

as I can not access the reader all the time as it is located in another building its a bit complicated to do "hands-on-stuff"

I used the volkszaehler rpi standard image.
I also tried to communicate and setup via "minicom" - but after doing this, there is silence on "cat /dev/ttyUSB0" and "cat /dev/ttyUSB0 | od -tx1"

I really don't now how to continue - any hints?

J-A-U, 2021-02-24:
> I think this is the wrong place 
So it is!
Everybody who likes to help please go to discussions at PV-Forum or Mailingslist.

closed 2021-02-24 by J-A-U

mr-gosh, 2021-02-24:
Danke - das Forum ist mir neu gewesen
Also falls wer auf die Frage stößt - weitere Kommunikation dazu hier:
https://www.photovoltaikforum.com/thread/152926-can-t-get-iskra-mt631-to-talk-with-my-elv-usb-iec-reader/