volkszaehler/vzlogger #457: make travis build less wasteful

r00t-, 2021-03-11:
i was already thinking it's kind of wasteful to, for example,
build the rather large "leptonica" library for the OCR meter from source three times, it's larger than vzlogger itself.

and by now it seems that with my recent commit activity, all our travis-ci credits were used up within a few days?

andig, 2021-03-11:
TravisCI ist mausetot als Produkt (von VC aufgekauft). Wir müssten z.B. zu GH actions umziehen- in anderen Projekten funktioniert das ganz gut. Diesmal fehlt mir leider die Zeit, den Umzug zu machen :(. volkszaehler.org ist umgestellt für build/lint, leider fehlen da aber auch noch all- unendlich wichtigen- Unit Tests :(

andig, 2021-03-11:
PS @r00t- Klasse dass Du so aufräumst, vielen Dank!

r00t-, 2021-03-14:
this took 40 me minutes: #460

i will not implement testing of meterOCR, as noted above.
(unless there is a binary package of leptonica.)

r00t-, 2021-03-15:
guess what, a few days after switching to github actions, we have used up our free quota there aswell.

https://docs.github.com/en/github/setting-up-and-managing-billing-and-payments-on-github/about-billing-for-github-actions#about-billing-for-github-actions
> Minutes reset every month, while storage usage does not.
> Product	Storage	Minutes (per month)
> GitHub Free	500 MB	2,000

r00t-, 2021-03-15:
@mbehr1:
so much for adding the extra compiler versions back...

r00t-, 2021-03-16:
looks like actions took a break only and is back now... weird.

i could not find any interface at github that would display a status of a project's actions or remaining budget.

r00t-, 2021-03-30:
seems to be working ok now.
for any new features deemed important,
just make MRs

PR #461 (2021-03-14): github actions
fix #457
closed 2021-03-30 by r00t-