volkszaehler/vzlogger #468: MQTT Error Codes

Sieboldianus, 2021-03-23:
Is there any guidance towards error codes of vzlogger mqtt implementation? I only found named errors on [the docs](https://mosquitto.org/api/files/mosquitto-h.html#mosquitto_publish), but vzlogger gives me numbers:

```
[Mar 23 22:28:50][mqtt] mosquitto_publish returned 4
[Mar 23 22:28:50][mqtt] mosquitto_reconnect succeeded
[Mar 23 22:28:50][mqtt] disconnect_callback called, res=6
[Mar 23 22:28:50][mqtt] mosquitto_loop returned 6. trying reconnect
[Mar 23 22:28:51][mqtt] publish vzlogger/data/chn0/raw=438.000000
[Mar 23 22:28:51][mqtt] mosquitto_publish returned 4
[Mar 23 22:28:51][mqtt] publish vzlogger/data/chn1/raw=998546.400000
[Mar 23 22:28:51][mqtt] mosquitto_publish returned 4
[Mar 23 22:28:51][mqtt] publish vzlogger/data/chn2/raw=1864560.800000
[Mar 23 22:28:51][mqtt] mosquitto_publish returned 4
[Mar 23 22:28:51][mqtt] mosquitto_reconnect succeeded
[Mar 23 22:28:51][mqtt] disconnect_callback called, res=6
[Mar 23 22:28:51][mqtt] mosquitto_loop returned 6. trying reconnect
```

I tried several config versions for `/etc/mosquitto/conf.d/default.conf`:

<details><summary>
1. No certs, no sockets
</summary>

```
allow_anonymous false
password_file /etc/mosquitto/passwd
listener 1883
protocol websockets
```
</details>

<details><summary>
2. With socket support
</summary>

```
allow_anonymous false
password_file /etc/mosquitto/passwd
listener 1883
```

</details>

<details><summary>
3. With certs (no socket)
</summary>

```
listener 8883
certfile /etc/nginx/ssl/wildcard.local.mytld.com.fullchain
keyfile /etc/nginx/ssl/wildcard.local.mytld.com.key
```

</details>

<details><summary>
4. With certs and socket support
</summary>

```
listener 8083
protocol websockets
cafile /etc/nginx/ssl/wildcard.local.mytld.com.ca
certfile /etc/nginx/ssl/wildcard.local.mytld.com.crt
keyfile /etc/nginx/ssl/wildcard.local.mytld.com.key
```

</details>

All of this with adjusted settings in MQTT of `/etc/vzlogger.conf`.

To no avail. The messages from vzlogger do not reach the MQTT Broker.

Depending on my settings above (with/without socket), I get slighlty different error numbers:

<details><summary>This is with socket:</summary>

```
[Mar 23 22:28:45][mqtt] mosquitto_publish announce vzlogger/data/chn0/id returned 4
[Mar 23 22:28:45][mqtt] mosquitto_publish announce vzlogger/data/chn0/uuid returned 4
[Mar 23 22:28:45][mqtt] mosquitto_publish announce vzlogger/data/chn1/id returned 4
[Mar 23 22:28:45][mqtt] mosquitto_publish announce vzlogger/data/chn1/uuid returned 4
[Mar 23 22:28:45][mqtt] mosquitto_publish announce vzlogger/data/chn2/id returned 4
[Mar 23 22:28:45][mqtt] mosquitto_publish announce vzlogger/data/chn2/uuid returned 4
[Mar 23 22:28:45][mqtt] mosquitto_reconnect succeeded
[Mar 23 22:28:45][mqtt] disconnect_callback called, res=6
[Mar 23 22:28:45][mqtt] mosquitto_loop returned 6. trying reconnect
[Mar 23 22:28:46][mqtt] mosquitto_publish announce vzlogger/data/chn0/id returned 4
[Mar 23 22:28:46][mqtt] mosquitto_publish announce vzlogger/data/chn0/uuid returned 4
[Mar 23 22:28:46][mqtt] publish vzlogger/data/chn0/raw=438.000000
[Mar 23 22:28:46][mqtt] mosquitto_publish returned 4
[Mar 23 22:28:46][mqtt] mosquitto_publish announce vzlogger/data/chn1/id returned 4
[Mar 23 22:28:46][mqtt] mosquitto_publish announce vzlogger/data/chn1/uuid returned 4
[Mar 23 22:28:46][mqtt] publish vzlogger/data/chn1/raw=998545.800000
[Mar 23 22:28:46][mqtt] mosquitto_publish returned 4
[Mar 23 22:28:46][mqtt] mosquitto_publish announce vzlogger/data/chn2/id returned 4
[Mar 23 22:28:46][mqtt] mosquitto_publish announce vzlogger/data/chn2/uuid returned 4
[Mar 23 22:28:46][mqtt] publish vzlogger/data/chn2/raw=1864560.800000
```

</details>

<details><summary>This is with certs:
</summary>

```
[Mar 23 22:18:28][mqtt] mosquitto_publish returned 4
[Mar 23 22:18:28][mqtt] mosquitto_reconnect succeeded
[Mar 23 22:18:28][mqtt] disconnect_callback called, res=6
[Mar 23 22:18:28][mqtt] mosquitto_loop returned 6. trying reconnect
[Mar 23 22:18:29][mqtt] publish vzlogger/data/chn0/raw=569.000000
[Mar 23 22:18:29][mqtt] mosquitto_publish returned 4
[Mar 23 22:18:29][mqtt] publish vzlogger/data/chn1/raw=998453.200000
[Mar 23 22:18:29][mqtt] mosquitto_publish returned 4
[Mar 23 22:18:29][mqtt] publish vzlogger/data/chn2/raw=1864560.800000
[Mar 23 22:18:29][mqtt] mosquitto_publish returned 4
[Mar 23 22:18:29][mqtt] mosquitto_reconnect succeeded
[Mar 23 22:18:29][mqtt] disconnect_callback called, res=6
```

</details>

<details><summar>This was with certs and web socket support</summary>

```
[Mar 23 22:04:23][mqtt] publish vzlogger/data/chn1/raw=998327.200000
[Mar 23 22:04:23][mqtt] publish vzlogger/data/chn2/raw=1864560.800000
[Mar 23 22:04:24][mqtt] publish vzlogger/data/chn0/raw=558.000000
[Mar 23 22:04:24][mqtt] publish vzlogger/data/chn1/raw=998327.400000
[Mar 23 22:04:24][mqtt] publish vzlogger/data/chn2/raw=1864560.800000
[Mar 23 22:04:25][mqtt] publish vzlogger/data/chn0/raw=558.000000
[Mar 23 22:04:25][mqtt] publish vzlogger/data/chn1/raw=998327.400000
[Mar 23 22:04:25][mqtt] publish vzlogger/data/chn2/raw=1864560.800000
[Mar 23 22:04:26][mqtt] publish vzlogger/data/chn0/raw=557.000000
[Mar 23 22:04:26][mqtt] publish vzlogger/data/chn1/raw=998327.600000
[Mar 23 22:04:26][mqtt] publish vzlogger/data/chn2/raw=1864560.800000
[Mar 23 22:04:27][mqtt] publish vzlogger/data/chn0/raw=557.000000
[Mar 23 22:04:27][mqtt] publish vzlogger/data/chn1/raw=998327.800000
[Mar 23 22:04:27][mqtt] publish vzlogger/data/chn2/raw=1864560.800000
[Mar 23 22:04:28][mqtt] publish vzlogger/data/chn0/raw=557.000000
[Mar 23 22:04:28][mqtt] publish vzlogger/data/chn1/raw=998328.000000
[Mar 23 22:04:28][mqtt] publish vzlogger/data/chn2/raw=1864560.800000
[Mar 23 22:04:29][mqtt] publish vzlogger/data/chn0/raw=558.000000
[Mar 23 22:04:29][mqtt] publish vzlogger/data/chn1/raw=998328.000000
[Mar 23 22:04:29][mqtt] publish vzlogger/data/chn2/raw=1864560.800000
[Mar 23 22:04:30][mqtt] publish vzlogger/data/chn0/raw=558.000000
[Mar 23 22:04:30][mqtt] publish vzlogger/data/chn1/raw=998328.200000
[Mar 23 22:04:30][mqtt] publish vzlogger/data/chn2/raw=1864560.800000
```

No error messages, but I also did not see the data coming in at the MQTT Broker.

</details>

r00t-, 2021-03-24:
this lists the error codes returned by `mosquitto_publish()`:
https://mosquitto.org/api/files/mosquitto-h.html#mosquitto_publish
and this lists the numeric values:
https://github.com/eclipse/mosquitto/blob/master/include/mosquitto.h#L87
```
4 == MOSQ_ERR_NO_CONN 
```
vzlogger should probably decode those codes for the user,
ideally the library contains a function for that.

Sieboldianus, 2021-03-24:
Great, thanks! So the MOSQ_ERR Codes and the List of names are ordered in the docs. This helps!

Beyond this issue, I was also  #looking for hints to my issue. I am posting here my configs, if no one can see any obvious mistake, I might post this in a consumer forum, to see if I can get somewhere:

<details><summary><code>/etc/vzlogger.conf</code></summary>

```conf
/**
 * vzlogger configuration
 *
 */

{
    // General settings
    "daemon": true,        // run periodically
    "verbosity": 15,         // log verbosity (0=log_alert, 1=log_error, 3=log_warning, 5=log_info, 10=log_debug, 15=log_finest)
    "log": "/var/log/vzlogger.log", // log file, optional
    "retry": 30,            // http retry delay in  #seconds

    // Build-in HTTP server
    "local": {
        "enabled": false,   // enable local HTTPd for serving live readings
        "port": 8080,       // TCP port for local HTTPd
        "index": true,      // provide index listing of available channels if no UUID was requested
        "timeout": 30,      // timeout for long polling comet requests in seconds (0 disables comet)
        "buffer": -1        // HTTPd buffer configuration for serving readings, default -1
                            //   >0: number of seconds of readings to serve
                            //   <0: number of tuples to server per channel (e.g. -3 will serve 3 tuples)
    },
    // realtime notification settings
    "push": [
        {
            "url": "http://127.0.0.1:5582"  // notification destination, e.g. frontend push-server
        }
    ],

    // mqtt client support (if ENABLE_MQTT set at cmake generation)
    "mqtt": {
        "enabled": true,  // enable mqtt client. needs host and port as well
        "host": "mqtt.local.mytld.com", // mqtt server addr
        "port": 8883, // 1883 for unencrypted, 8883 enc, 8884 enc cert needed,
        "cafile": "/etc/nginx/ssl/wildcard.local.mytld.com.ca", // optional file with server CA
        "capath": "", // optional path for server CAs. see mosquitto.conf. Specify only cafile or capath
        "certfile": "/etc/nginx/ssl/wildcard.local.mytld.com.crt", // optional file for your client certificate (e.g. client.crt)
        "keyfile": "/etc/nginx/ssl/wildcard.local.mytld.com.key", // optional path for your client certficate private key (e.g. client.key)
        "keypass": "", // optional password for your private key
        "keepalive": 30, // optional keepalive in seconds.
        "topic": "vzlogger/data", // optional topic dont use $ at start and no / at end
        "user": "smartmeter", // optional user name for the mqtt server
        "pass": "...", // optional password for the mqtt server
        "retain": false, // optional use retain message flag
        "rawAndAgg": false, // optional publish raw values even if agg mode is used
    },

    // Meter configuration
    "meters": [
         {
         }
    ],
    ...
}
```

</details>

<details><summary><code>/etc/mosquitto/conf.d/default.conf</code></summary>

```conf
allow_anonymous false
password_file /etc/mosquitto/passwd
listener 1883 localhost

listener 8883
certfile /etc/nginx/ssl/wildcard.local.mytld.com.fullchain
keyfile /etc/nginx/ssl/wildcard.local.mytld.com.key

```

</details>

<details><summary><code>/etc/telegraf/telegraf.conf</code></summary>

```conf
[global_tags]
[agent]
  interval = "10s"
  round_interval = true
  metric_batch_size = 1000
  metric_buffer_limit = 10000
  collection_jitter = "0s"
  flush_interval = "10s"
  flush_jitter = "0s"
  precision = ""

  debug = true
  hostname = ""
  omit_hostname = false

[[inputs.mqtt_consumer]]
servers = ["tcp://localhost:1883"]
topics = [
    "vzlogger/#",
]
qos = 0
connection_timeout = "30s"
persistent_session = true
client_id = "influx_telegraf"
username = "telegraf"
password = "..."
data_format = "value"
```

</details>

Sieboldianus, 2021-03-24:
I think I isolated the error.

Using:
```
allow_anonymous true
```

in `/etc/mosquitto/conf.d/default.conf`, the vzlogger MQTT messages arrive:

```
tail -f /var/log/mosquitto/mosquitto.log
```
```
1616568634: Received PUBLISH from vzlogger_4467 (d0, q0, r0, m0, 'vzlogger/data/chn2/raw', ... (14 bytes))
1616568635: Received PUBLISH from vzlogger_4467 (d0, q0, r0, m0, 'vzlogger/data/chn0/raw', ... (12 bytes))
1616568635: Received PUBLISH from vzlogger_4467 (d0, q0, r0, m0, 'vzlogger/data/chn1/raw', ... (14 bytes))
1616568635: Received PUBLISH from vzlogger_4467 (d0, q0, r0, m0, 'vzlogger/data/chn2/raw', ... (14 bytes))
1616568636: Received PUBLISH from vzlogger_4467 (d0, q0, r0, m0, 'vzlogger/data/chn0/raw', ... (12 bytes))
```
```
tail -f /var/log/vzlogger.log | grep mqtt
```
```
[Mar 24 07:50:39][mqtt] publish vzlogger/data/chn1/raw=1003517.800000
[Mar 24 07:50:39][mqtt] publish vzlogger/data/chn2/raw=1865034.000000
[Mar 24 07:50:40][mqtt] publish vzlogger/data/chn0/raw=-1941.000000
[Mar 24 07:50:40][mqtt] publish vzlogger/data/chn1/raw=1003517.800000
[Mar 24 07:50:40][mqtt] publish vzlogger/data/chn2/raw=1865034.600000
[Mar 24 07:50:41][mqtt] publish vzlogger/data/chn0/raw=-1942.000000
[Mar 24 07:50:41][mqtt] publish vzlogger/data/chn1/raw=1003517.800000
[Mar 24 07:50:41][mqtt] publish vzlogger/data/chn2/raw=1865035.200000
```

Apparently, it has something to do with the user auth. I checked passwords several times and they are ok. I tested connecting locally with those credentials to the MQTT host, and it worked. Clueless as to now, but it does work for now..

Sieboldianus, 2021-03-24:
With the anonymous login it works. I also had to set the following telegraf configs:

```
[[inputs.mqtt_consumer]]
servers = ["tcp://localhost:1883"]
topics = [
    "vzlogger/#",
]
qos = 0
connection_timeout = "30s"
client_id = "influx_telegraf"
data_format = "value"
data_type = "float"
```

Not the `data_type = "float"`, which was the final issue that prevented values to flow to influxdb2.

r00t-, 2021-03-25:
to debug your authentication issues...
we should enable detailed logging of the http requests (via curl debug option) in vzlogger.

alternatively, point vzlogger at a tcp listener (`nc -vlp 1884` or such)
to intercept the data vzlogger is sending,
and compare it with a working request from another utility.

Sieboldianus, 2021-03-25:
I will try and report.. many thanks!

Sieboldianus, 2021-03-25:
Well, it was a user error.

In most guides for setting up MQTT broker, the following command is used to add use/pw:
```
sudo mosquitto_passwd -c /etc/mosquitto/passwd smartmeter
```

Of course, I did this twice:
- to add the `smartmeter` user
- and to add the `telegraf` user

What they don't tell you is that `-c` flag silently overwrites the passwd file, deleting the first user.

I found out through following the `nc` rabbit hole - thanks!

Everything looked fine, wich is why I finally checked the `/etc/mosquitto/passwd`, to find it contained only one user.

Start netcat:
```
nc -vlp 1884
```

.. and change vzlogger.conf to use `1884` on the RPi, I get from nc:

```
connect to [192.168.60.10] from (UNKNOWN) [192.168.60.16] 52912
<MQTT▒vzlogger_14807
smartmeterlk8wwefewf7uZF7NoZLk0%vzlogger/data/chn0/id1-0:16.7.0*2550=vzlogger/data/chn0/uuiddcdbec30-87f6-11eb-bf60-e5d3715178680$vzlogger/data/chn0/raw-6330.0000000$vzlogger/data/chn1/id1-0:1.8.0*2550=vzlogger/data/chn1/uuidbd6c93b0-87f4-11eb-9599-a1e7fb6c47040&vzlogger/data/chn1/raw1011852.8000000$vzlogger/data/chn2/id1-0:2.8.0*2550=vzlogger/data/chn2/uuid2ebb8470-87f7-11eb-8877-3198b438162f0&vzlogger/data/chn2/raw1999996.6000000$v
```

- `smartmeter` here is the MQTT user
- `lk8wwefewf7uZF7NoZLk` is MQTT user's password
- not sure what `0%` is

And in vzlogger logs:
```
[Mar 25 13:46:04][mqtt] mosquitto_loop returned 4. trying reconnect
[Mar 25 13:46:05][mqtt] mosquitto_reconnect returned 14
[Mar 25 13:46:05][mqtt] mosquitto_loop returned 4. trying reconnect
[Mar 25 13:46:06][mqtt] mosquitto_reconnect returned 14
[Mar 25 13:46:06][mqtt] mosquitto_loop returned 4. trying reconnect
[Mar 25 13:46:07][mqtt] mosquitto_reconnect returned 14
[Mar 25 13:46:07][mqtt] mosquitto_loop returned 4. trying reconnect
```

Test the same nc connection with mosquitto_pub:
```
mosquitto_pub -h 192.168.60.10 -t test -m "hello world" -p 1884 -u "smartmeter" -P "lk8wwefewf7uZF7NoZLk"
```

```
connect to [192.168.60.10] from 192.168.60.10 [192.168.60.10] 52172
?MQTT▒<mosqpub|17462-iot
smartmeterlk8wwefewf7uZF7NoZLk
```

Many thanks! And many apologies for this unecessary issue report.

I still leave it open if you want to reference it when error codes are decoded. Feel free to close it though.

r00t-, 2021-03-25:
> What they don't tell you is that -c flag silently overwrites the passwd file, deleting the first user.

so the root cause is that `mosquitto_passwd` copies the syntax of `htpasswd`, which has the bad design decision that `-c` means "create password file", not "create user"...
(and adding a user to a nonexistent file does not implicitly create the file but fails)

i will try to create an MR for adding 'mosquitto_strerror()', and you can then help by testing it.

PR #469 (2022-04-20): [mqtt] always use mosquitto_strerror() instead of printing raw error codes
fix #468
closed 2022-04-20 by J-A-U