volkszaehler/vzlogger #471: Broken second meter

StefanSchoof, 2021-03-29:
I wanted to update to the latest vzlogger to use the new foreground option. But with the latest version b2782d81da441ce51864e4842b4091e749ec4bd3 only my first meter works. It looks like second meter is just ignored.

Then I tried 77872669157f01739e50fd550df99f448689f8e0 and it works like previously. In the volkszahler users mailing list there was a similar report (https://demo.volkszaehler.org/pipermail/volkszaehler-users/2021-March/015555.html)

r00t-, 2021-03-29:
possible that one of the commits in between caused a regression,
but nothing comes to mind immediately.
would you mind doing some more testing with commits in between?
i don't have any smartmeters around, and can't reproduce with two random meters.
(or maybe try your confg (which you did not post) with two random meters instead of the real ones, and see if that reproduces the issue?)

StefanSchoof, 2021-03-29:
I will try to do some more digging, but it may take some time.

StefanSchoof, 2021-03-31:
I tried to reproduce this error and was not able. With a rebuild and my current config it works. So I can not say if I did the error in testing in the first place or if there are some other conditions trigger this behavior.

closed 2021-03-31 by StefanSchoof

r00t-, 2021-03-31:
as said, i can not imagine how any of the commits between b2782d8 and 7787266 could possibly cause such a bug, they are all unrelated to meter handling.
but still strange to also have that report on the ML.

@StefanSchoof: could you post a link to this ticket in that ML thread, i don't follow the -users list.

StefanSchoof, 2021-03-31:
@r00t- Here the link https://demo.volkszaehler.org/pipermail/volkszaehler-users/2021-March/015555.html

r00t-, 2021-03-31:
@StefanSchoof:
ich sprach davon, einen link in gegenrichtung zu erstellen,
falls sich auf der mailingliste neue erkentnisse ergeben.
habe das jetzt doch selbst erledigt:
https://demo.volkszaehler.org/pipermail/volkszaehler-users/2021-April/015575.html
(wenn auch eine stunde zu spaet, so dass der post es im listenarchiv nichtmehr in den gleichen thread geschafft hat.)