volkszaehler/vzlogger #481: post-processing readings inside vzlogger?

r00t-, 2021-04-02:
it appears that it would sometimes be desirable to have some ability to manipulate readings inside vzlogger before sending them off to an API.

for example: https://github.com/volkszaehler/vzlogger/issues/431#issuecomment-812672249
it's desired to log negative (delivery) and positive (consumption) power reading into separate channels.
in the #431 case, we could (should!) also provide the sml "energy direction" status bit as a separate channel,
and could then use generic logic to apply it as the sign of the power channel, or log power to a different output channel depending on it.
there might also be S0 meters that output a direction signal in addition to the impulses, that could also be processed that way.
building generic functionality instead of hardcoding it in specific places (same as #445).
(if you have further examples, please comment)

volkszaehler.org has gained "virtual channells" in https://github.com/volkszaehler/volkszaehler.org/pull/729 ,
which is a similar feature.
but we might want to:
* manipulate the data _before_ logging it
* use this with other APIs than volkszaehler.org

such a featute could either be added to the vzlogger core (like aggregation),

or, which seems somehow sensible to me at the moment,
as a module that exposes both an API _and_ a meter, "loopback"-style,
so that readings from any meters can be sent to it with the existing config syntax
(which would also provide a way to pass processing options),
and it could process them and output the results as readings for loggint to another "real" API.
(no idea if i'll get to implement this or any of it, as i have no use for it personally.)

J-A-U, 2021-04-02:
> volkszaehler.org has gained "virtual channells" in volkszaehler/volkszaehler.org#729 ,
I'm already trying to wrap my brain around how this could be applied to e.g. separated channels. Merging them together how ever the user wants them to be. Or an flag [-1] to mathematically apply to the data. Looking for a solution to keep vzlogger-config easy.

It's pretty complicated to get the messages across. Maybe it heavily collides with user-expectations.

And my thoughts only apply to VZ...

r00t-, 2021-04-02:
@J-A-U:
it's a typical tradeoff, you can't have complex features with an "easy" configuration.
this won't happen before we have strong AI than can actually understand what the user wants from any nonsensical inputs they give.

r00t-, 2021-04-02:
quoting @J-A-U from https://github.com/volkszaehler/vzlogger/issues/431#issuecomment-812687394

> This may be an option with different states: only positive values, only negatives and both.
> But it would be necessary to config this in [channel] instead of [meter].

we could ofcourse provide a hardcoded sub-set of transformations (and thus sub-set of the feature suggested above) as channel-properties,
like for this case, we would need combinations of `min()`, `abs()` and `max()`, like:
`abs(min(value,0))` to get only the export part of the energy readings,
and `max(value,0)) to get only the consumption part.
(incidentially that's what `aggmode=max` did before #... was fixed ;) )

at best we could pull in some library to allow arbitrary expressions of that style.

but i'm reluctant to implement such a thing, because supporting only transformations within one channel, while it might solve SOME use-cases, is very limited, and sooner or later somebody will come along and ask for something more complex.