volkszaehler/vzlogger #484: QUESTION: client id dynamic

Talkabout, 2021-04-06:
Hi,

I am using vzlogger since a few days now but facing an issue caused by the dynamic client id. Currently the transferred topics have a format like:

vzlogger_PID:vzlogger/data/....

The problem is that my home automation system is expecting the client id to be the same for the same device. I have adjusted the vzlogger code to allow config definition of client id and only in case it is not set the dynamic one is generated. Is this something that makes sense for the standard implementation? If so I'd create a pull request.

Bye

r00t-, 2021-09-16:
@Talkabout: please do create a MR

Talkabout, 2021-09-17:
PR added [498](https://github.com/volkszaehler/vzlogger/pull/498)

closed 2021-09-17 by J-A-U