volkszaehler/vzlogger #487: influxdb api allow naming of field(s)

antonmeyer, 2021-04-12:
it is a feature request:
allow (re)naming of "value" of fields in a channel.
e.g. "temp" or "hum".
Not a big issue as long you have only 1 value per channel,
but it is closer to the influx api

r00t-, 2021-04-14:
so this is hardcoded here:
https://github.com/volkszaehler/vzlogger/blob/fe937770e44f648e8950e414455caee91b75b5bd/src/api/InfluxDB.cpp

the code has access to the Channel object in `channel()`,
for example it retrieves the channel's UUID a few lines above:
https://github.com/volkszaehler/vzlogger/blob/master/src/api/InfluxDB.cpp#L292

any property of it that was defined in vzlogger.conf can be accessed in `channel()->options()`,
should be no problem to retrieve a "influxdb_field_name" there.
(would not be the most efficient thing to parse the options while iterating there, should be cached somehow.)

r00t-, 2021-04-14:
@antonmeyer:

are you aware that you can already specify tags for each channel?
that is what i'm using, and while it's weird that the field is always called "value",
the tag works just fine to identify the reading.

```
{
 "api": "influxdb",
 "send_uuid": false,
 "measurement_name": "vzlogger",
 "tags": "channel=kitchen,identifier=Power",
 "identifier": "Power"
}
```

antonmeyer, 2021-04-14:
"... always called "value", 
A: when you use the querybuilder you might endup in having the column named "value" for different kind of sensors. having meaningful names to the fields makes it just easier in influxdb to mix different kind of series in 1 dashboard / cell. For sure this is not a show stopper, as you can solve it in influxdb with renaming. it is just nice to have.

matzep2002, 2022-08-20:
> "... always called "value",
> A: when you use the querybuilder you might endup in having the column named "value" for different kind of sensors. having meaningful names to the fields makes it just easier in influxdb to mix different kind of series in 1 dashboard / cell. For sure this is not a show stopper, as you can solve it in influxdb with renaming. it is just nice to have. And as the source code is well structured I already add that feature, just need to test it.

Hello I am also interested in the function. Did you install this in yours? thank you greetings Matze

antonmeyer, 2022-08-25:
Hi Matze,

I moved away from volkszähler and wrote my own SML2InfluxDB.
By that the serial channel is complete flexible by decoding the meter ID from
the SML an uses that as tag.
I think volkszähler does not support this concept.
down side of my SMl2InfluxDB is, that the reported values are hard coded
for actual power and overall energy.
works for me.
and works on an ESP32.

cheers rene