volkszaehler/vzlogger #488: timestamp 10 years in the future

lapawa, 2021-04-15:
I'm running vzlogger on a Raspberry Pi 1 Model B Rev 2 with the default 32Bit Raspbian Linux on it.
The hardware is quite old but fits perfect for this task.
I've connected a serial optical device via USB to meter with the SML protocol.
Values are send to MSQTT and five channels with OBIS identifiers to an influxdb.
In this constellation vzlogger calculates and sends a wrong timestamp. 

Some messages from the vzlogger.log file:
[Apr 15 21:02:50][mtr0] Got 5 new readings from meter:
[Apr 15 21:02:50][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=418906.10 ts=1930876000
[Apr 15 21:02:50][mtr0] Reading: id=1-0:36.7.0*255/ObisIdentifier:1-0:36.7.0*255 value=64.00 ts=1930876000
[Apr 15 21:02:50][mtr0] Reading: id=1-0:56.7.0*255/ObisIdentifier:1-0:56.7.0*255 value=65.00 ts=1930876000
[Apr 15 21:02:50][mtr0] Reading: id=1-0:76.7.0*255/ObisIdentifier:1-0:76.7.0*255 value=234.00 ts=1930876000
[Apr 15 21:02:50][mtr0] Reading: id=1-0:16.7.0*255/ObisIdentifier:1-0:16.7.0*255 value=364.00 ts=1930876000
[Apr 15 21:02:50][chn0] Adding reading to queue (value=418906.10 ts=1930876000)
[Apr 15 21:02:50][mqtt] publish vzlogger/data/chn0/raw={ "timestamp": 1930876000, "value": 418906.10000000003 }
[Apr 15 21:02:50][chn1] Adding reading to queue (value=64.00 ts=1930876000)
[Apr 15 21:02:50][mqtt] publish vzlogger/data/chn1/raw={ "timestamp": 1930876000, "value": 64 }
[Apr 15 21:02:50][chn2] Adding reading to queue (value=65.00 ts=1930876000)
[Apr 15 21:02:50][mqtt] publish vzlogger/data/chn2/raw={ "timestamp": 1930876000, "value": 65 }
[Apr 15 21:02:50][chn3] Adding reading to queue (value=234.00 ts=1930876000)
[Apr 15 21:02:50][mqtt] publish vzlogger/data/chn3/raw={ "timestamp": 1930876000, "value": 234 }
[Apr 15 21:02:50][chn4] Adding reading to queue (value=364.00 ts=1930876000)
[Apr 15 21:02:50][mqtt] publish vzlogger/data/chn4/raw={ "timestamp": 1930876000, "value": 364 }

But 1930876000 is ten years in the future!

$ date --date='@1930876000'
Mon 10 Mar 2031 03:26:40 AM CET

Systemtime on the other hand is correct:
$ date +"%s"
1618512610
$ date --date='@1618512610'
Thu 15 Apr 2021 08:50:10 PM CEST

Perhaps there's something wrong with the 64bit time_ms() inline calculation in 
Reading.hpp:156:        int64_t time_ms() const { return ((int64_t)_time.tv_sec) * 1e3 + (_time.tv_usec / 1e3); };

I've compiled tvlogger from this repo:
commit fe937770e44f648e8950e414455caee91b75b5bd (HEAD -> master, origin/master, origin/HEAD)

And with this compiler:
 $ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/8/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 8.3.0-6+rpi1' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-8 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-werror --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1) 

Could you please verify this behaviour? 

Many thanks

J-A-U, 2021-04-15:
Vzlogger usually uses the time the meter provides. Unfortunately the re not always set up the right way.
Solution for this is the parameter `"use_local_time": true,`
https://wiki.volkszaehler.org/software/controller/vzlogger/vzlogger_conf_parameter#sml

r00t-, 2021-04-15:
@lapawa:
as a bug report this is useless if you don't provide a dump of the data from your meter,
so we can check what vzlogger is parsing there.
see https://github.com/devZer0/libsml-testing/blob/master/README.md

lapawa, 2021-04-16:
> Vzlogger usually uses the time the meter provides. Unfortunately they are not always set up the right way.
> Solution for this is the parameter `"use_local_time": true,`
>  #https://wiki.volkszaehler.org/software/controller/vzlogger/vzlogger_conf_parameter#sml

It's getting even worse, when I switch the configuration to  `"use_local_time": true,`
The timesamp became negativ.

```[Apr 16 17:02:58][mtr0] Got 5 new readings from meter:
[Apr 16 17:02:58][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=428454.30 ts=-617292592
[Apr 16 17:02:58][mtr0] Reading: id=1-0:36.7.0*255/ObisIdentifier:1-0:36.7.0*255 value=64.00 ts=-617292592
[Apr 16 17:02:58][mtr0] Reading: id=1-0:56.7.0*255/ObisIdentifier:1-0:56.7.0*255 value=70.00 ts=-617292592
[Apr 16 17:02:58][mtr0] Reading: id=1-0:76.7.0*255/ObisIdentifier:1-0:76.7.0*255 value=161.00 ts=-617292592
[Apr 16 17:02:58][mtr0] Reading: id=1-0:16.7.0*255/ObisIdentifier:1-0:16.7.0*255 value=296.00 ts=-617292592
[Apr 16 17:02:58][chn0] Adding reading to queue (value=428454.30 ts=-617292592)
[Apr 16 17:02:58][mqtt] publish vzlogger/data/chn0/raw={ "timestamp": -617292592, "value": 428454.30000000005 }
[Apr 16 17:02:58][chn1] Adding reading to queue (value=64.00 ts=-617292592)
[Apr 16 17:02:58][mqtt] publish vzlogger/data/chn1/raw={ "timestamp": -617292592, "value": 64 }
[Apr 16 17:02:58][chn2] Adding reading to queue (value=70.00 ts=-617292592)
[Apr 16 17:02:58][mqtt] publish vzlogger/data/chn2/raw={ "timestamp": -617292592, "value": 70 }
[Apr 16 17:02:58][chn3] Adding reading to queue (value=161.00 ts=-617292592)
[Apr 16 17:02:58][mqtt] publish vzlogger/data/chn3/raw={ "timestamp": -617292592, "value": 161 }
[Apr 16 17:02:58][chn4] Adding reading to queue (value=296.00 ts=-617292592)
[Apr 16 17:02:58][mqtt] publish vzlogger/data/chn4/raw={ "timestamp": -617292592, "value": 296 }
```

r00t-, 2021-04-17:
@lapawa:
with the dumps you uploaded, i can reproduce the issue,
in so far as that the meter sends useless time values.

but note that:
 * it should definitely work if you set use_local_time, because then the time sent by the meter is simply ignored.
   i can not reproduce those negative time values you are seeing, and have no idea how those could happen, specifically with use_local_time set. you should also power your config that procudes those readings.
 * also note that the invalid time values are far in the past, not in the future, as vzlogger sends millisecond-timestamps to the API - you need to strip the three trailing zeroes or divide by 1k.

as J-A-U already pointed out, most likely your meter simply does not have a correctly set clock.
most likely it's simply counting seconds since it was powered up.

check the difference between the timestamp in your original post,and the timestamp from your dump:
$ date -d @1930876
23 Jan 1970 09:21:16 AM CET
$ date -d @2005973
24 Jan 1970 06:12:53 AM CET

i would guess that your meter has been running for 23 days since it's installation or the the last time it had no power supplied to it.

r00t-, 2021-04-17:
i guess we should have vzlogger print an error or abort or automatically use local time,
if it receives timestamps from far in the past.
and maybe have an option to override that: `"i_have_a_time_machine:" true`

lapawa, 2021-04-18:
This makes absolutely sense. I missed to convert the timestamp from ms to seconds and the meter was freshly installed about three weeks ago. 

Unfortunately I'm unable to reproduce the negative timestamp.
The option `"use_local_time": true` solves the issue for me.

closed 2022-04-20 by J-A-U