volkszaehler/vzlogger #492: segfault since 4 weeks

andreasschulze, 2021-05-16:
building vzlogger this way:

```
cd /tmp/
git clone https://github.com/volkszaehler/vzlogger
cd /tmp/vzlogger/
docker build -t vzlogger .
```

now starting using this config:
```
# cat <<'EOF' > /tmp/vzlogger.conf
{
  "verbosity": 15,
  "local": {
    "enabled": true,
    "port": 1025,
    "index": true
  },
  "meters": [
    {
      "enabled": true,
      "protocol": "random",
      "interval": 299,
      "max": 40.5,
      "min": -20.5,
      "channel": {
        "uuid": "00000000-0000-0000-0000-000000000001",
        "identifier": "random",
        "api": "null"
      }
    }
  ]
}
EOF
```
```
$ docker run --rm -ti -v /tmp:/cfg --entrypoint /bin/bash vzlogger
root@e3b661874780:/# vzlogger -c /cfg/vzlogger.conf -f
Segmentation fault
```

this start to happen 4 weeks ago

andreasschulze, 2021-05-16:
clarification: the config file was indended only to run some regessiontests after build:
```
wget -q -O - localhost:1025
wget -q -O - localhost:1025 | jq -M -e '.generator' | grep vzlogger
wget -q -O - localhost:1025 | jq -M -e '.data[0].protocol' | grep random
```

wget should simply not fail to pass my simple tests

r00t-, 2021-05-16:
master with your example config crashes in `NilIdentifier::isEqual()`, probably caused by #422

andreasschulze, 2021-08-22:
@claashk your fix is still not included in https://github.com/volkszaehler/vzlogger, right? Do I've still to wait until that happen?

other option: what is wrong with the minimal configuration file? ist there anything I could/should change to avoid triggering the bug?

Thanks!

claashk, 2021-08-24:
I modified the associated https://github.com/claashk/vzlogger/tree/issue492 branch. The change to be merged to the master branch by a member of the core maintenance team, though. 

@r00t- : Can you do that or are there remaining open issues? 

There is nothing wrong with your config as far as I can tell.

andreasschulze, 2021-09-03:
I like to confirm commit https://github.com/volkszaehler/vzlogger/commit/a61a2457f33b8fe1eafed53e169afbbc0cf480a6 fixes the reported problem. vzlogger now work again for me. Thanks

This issue may be closed.

closed 2021-09-14 by r00t-