volkszaehler/vzlogger #494: Support of Kaifa MA309

tirolerstefan, 2021-05-26:
I have got a new smart meter "Kaifa MA309" which sends (encrypted) push messages via MBUS. 
I connected this to my Raspberry Pi using a USB serial cable and compiled vzlogger.

I have configured vzlogger this way:
```
{
  "retry": 1,
  "verbosity": 15,
  "log": "/var/log/vzlogger.log",
  "push": [],
  "meters": [
    {
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [],
      "protocol": "oms",
      "device": "/dev/ttyUSB0",
      "baudrate": 2400,
      "key": "*****************",
      "mbus_debug": true,
      "use_local_time": true,
    }
  ]
}
```

But when I start the `vzlogger`, the output shows only something like this:

```
...
[May 16 00:02:31][mtr0] Got 0 new readings from meter:
[May 16 00:02:32][mtr0] Got 0 new readings from meter:
[May 16 00:02:32][oms]  got valid mbus frame with len=0, type=1 control=6c controlinfo=0 address=64
[May 16 00:02:32][mtr0] Got 0 new readings from meter:
[May 16 00:02:32][mtr0] Got 0 new readings from meter:
...
```

or this

```
...
[May 23 11:03:08][mtr0] Got 0 new readings from meter:
[May 23 11:03:08][mtr0] Got 0 new readings from meter:
[May 23 11:03:08][oms]  got valid mbus frame with len=150, type=1 control=9b controlinfo=0 address=56
[May 23 11:03:08][oms]  wrong frame received. waiting for SND_NKE or SND_UD!
[May 23 11:03:08][mtr0] Got 0 new readings from meter:
[May 23 11:03:08][mtr0] Got 0 new readings from meter:
...
```

An example message (encrypted) looks like this:

```
0000000 c9 41 41 49 6b 80 ff fd 62 92 7a d2 73 ca fb 7e
0000010 fa 3a 72 d3 f5 aa eb f7 e3 5e f8 6a 61 d9 a5 b8
0000020 9b 6f 75 7b 8b a1 1e 8d 17 70 0b 6d 7e 18 ba ff
0000030 ac 3a bd 14 15 10 8c b5 ff 71 29 22 87 ce a9 73
0000040 38 77 69 19 20 1f c9 bc 1b 52 a4 b3 37 46 b2 92
0000050 f8 2e e5 97 c6 fb dd 33 1d b8 4a af 92 d1 94 c5
0000060 47 31 cf 6c 1b c0 cc 44 58 0d aa 55 4c 89 47 36
0000070 26 d9 fa aa 15 65 72 41 81 8f 72 3d ab 7f 1a e1
0000080 af 04 e5 5a 2b 11 ce 44 79 1b 2d b4 15 b0 32 eb
0000090 b1 43 2e 79 69 f6 15 02 aa 1d b6 a2 d5 33 cd bd
00000a0 51 7e d6 d1 92 ac ec ad ff 38 c2 a2 d9 82 ac 43
00000b0 22 4e 19 b9 a3 f5 a1 c8 7a 81 53 d1 24 6d b9 e0
00000c0 50 9b b6 3f 8a b5 b7 5c 61 44 47 91 f1 a7 9e 09
00000d0 30 d8 94 67 14 bd 39 c7 0f 21 3a 7b d8 0b 36 45
00000e0 39 76 63 d4 4d 59 8f 5e 09 84 6c 30 ef 68 e4 e8
00000f0 8e 6d dc b6 27 b1 80 89 a2 ab 92 28 4b a9 cd 2e
0000100 e6 b3 e5 18 60 6d 91 40 66 87 bb 3a 00 c9 6c ec
0000110 49 2d a0 af 62 d8 f7 a7 a5 3e 43 10 4c 93 cb c5
0000120 98 62 c9 8b 7d af 92 47 28 45 9b 32 b6 85 b5 2d
0000130 98 75 50 8c 91 90 61 82 a7 ca 08 53 76 5f ad 0f
0000140 1e 67 a3 03 15 6a c6 8e bd 5e 23 a7 35 83 dd 6a
0000150 a4 d6 d1 d9 52 40 b2 cb 35 1b fe 2b 87 ec 54 c1
0000160 b9 76 89 d5 4b bb 12 27 c4 75 33 b6 a9 d2 19 88
0000170 f4 8a a5 a0 25 93 6f 95 0d b8 ae 80 f7 d1 47 80
0000180 d1 42 9a 48 3c 56 c6 82 3a 00  
```

Documentation of the smart meter interface is here:
https://www.tinetz.at/fileadmin/user_upload/Kundenservice/pdf/Beschreibung_Kundenschnittstelle_Smart_Meter_TINETZ.pdf

Can anyone help me here?

baumi85, 2021-06-02:
I have the same challenge with the Tinetz Kaifa MA309, wired Mbus to USB converter on raspberry Pi.
Did you find a way to decrypt the data with the provided key?

tirolerstefan, 2021-06-02:
Unfortunate not, yet :-(.

tirolerstefan, 2021-06-03:
@mbehr1 ... do you have an idea how we could get that data decoded?

mbehr1, 2021-06-03:
@tirolerstefan according to spec it's
AES128-CBC
can you provide some test data (encrypted) and the corresponding key?

tirolerstefan, 2021-06-03:
> @tirolerstefan according to spec it's
> AES128-CBC
> can you provide some test data (encrypted) and the corresponding key?

Hi Matthias - thanks for answering! For an encrypted sample, please check the first post.
I will send you my AES key by email.

mdvlbg, 2021-09-16:
Hallo Stefan.

Habe nun auch einen Kaifa MA309 bekommen und bekomme ihn auch schon zum auslesen, nur der vzlogger will noch nicht das machen. hast du einen Tip für mich?

Den Code habe ich schon.

tirolerstefan, 2021-09-16:
Hi @mdvlbg , soweit ich weiß, wurde der Kaifa MA309 mit seinen beiden Telegrammen noch nicht implementiert.
Ich habe mir selbst geholfen und eine eigenes Programm zum Auslesen geschrieben, momentan mit Fokus auf die Einspeisung in Solarview.
Alle Details:
https://github.com/tirolerstefan/kaifa/