volkszaehler/vzlogger #496: Support specific clientid in configuration when using mqtt

roaroid, 2021-09-10:
When using mqtt to send the data to a mqtt broker it currently seems that the clientid used to connect to the mqtt broker is defined as `vzlogger_<pid>`

[https://github.com/volkszaehler/vzlogger/blob/a61a2457f33b8fe1eafed53e169afbbc0cf480a6/src/mqtt.cpp#L99](https://github.com/volkszaehler/vzlogger/blob/a61a2457f33b8fe1eafed53e169afbbc0cf480a6/src/mqtt.cpp#L99)

Is it possible to change this behaviour e.g. by adding a new attribute `clientid` to `vzlogger.conf` in the mqtt-section?
The actual behaviour could then be used as fallback if no `clientid` was specified in `vzlogger.conf`.

The background is that mosquitto broker has a security feature called `clientid_prefixes`:

```
# If set, only clients that have a matching prefix on their
# clientid will be allowed to connect to the broker. By default,
# all clients may connect.
# For example, setting "secure-" here would mean a client "secure-
# client" could connect but another with clientid "mqtt" couldn't.
clientid_prefixes secure-
```

When using this feature in mosquitto it is currently not possible to connect to the broker using vzlogger.

r00t-, 2021-09-16:
i guess adding `"vzlogger_"` to the `clientid_prefixes` is not a valid solution?

r00t-, 2021-09-16:
this is also a duplicate of https://github.com/volkszaehler/vzlogger/issues/484 , the author of which says they did already implement this.

r00t-, 2021-09-17:
@roaroid : 
can you build and try the fix in https://github.com/volkszaehler/vzlogger/pull/498 ?

roaroid, 2021-09-17:
I tested the fix provided in [#498](https://github.com/volkszaehler/vzlogger/pull/498). The fix works for me without any problems. Thank you!

closed 2021-09-17 by J-A-U