volkszaehler/vzlogger #497: aggregation of multiple meters

zinz-code, 2021-09-13:
Is it possible to force vzlogger to send multiple meters aggregated?
I have 4 meters in a[ cascade configuration](https://www.dgs.de/news/en-detail/050321-ue20-fuer-solar-dummies/)  sadly this means I have to do a lot of calculations for the values I want. As you can see this leads to quite a lot of calculation artifacts

[image]

https://www.dgs.de/news/en-detail/050321-ue20-fuer-solar-dummies/

r00t-, 2021-09-14:
this is essentially a duplicate of https://github.com/volkszaehler/vzlogger/issues/481 ,
where this is already being discussed in more detail.

r00t-, 2021-09-14:
note that if your meters send data at async/irregular intervals (specifically s0 meters) this can become a pretty hard problem.

zinz-code, 2021-09-15:
> 
> 
> this is somewhat a duplicate of #481 ,
> where options leading in this direction are already being discussed in more detail.

It certainly looks like one of the possibly use cases of that issue

> 
> 
> note that if your meters send data at async/irregular intervals (specifically s0 meters) this can become a pretty hard problem.

thankfully I don't have that Problem as these are all smart meters where I get the meter readings