volkszaehler/vzlogger #502: Mosquitto/MQTT with major version 2 won't start

ngoldack, 2021-10-10:
Hi all, 
when trying to configure vzlogger with mqtt I approched following error:

```
[Oct 10 23:42:29][mqtt] MqttClient::MqttClient called
[Oct 10 23:42:29][mqtt] using libmosquitto 2.0.11
[Oct 10 23:42:29][mqtt] Wrong libmosquitto major version! 2 vs. expected 1! Stopped.
[Oct 10 23:42:29][mqtt] ~MqttClient called
[Oct 10 23:42:29][mqtt] mqtt client not configured. stopped.
```

I'm using a mosquitto broker with the newest version of 2.0.11.

I've inspected the source code and found that in `mqtt.cpp` [it checks for the major version 1](https://github.com/volkszaehler/vzlogger/blob/49240dfdd5300d497cf7e67352037bc2db888926/src/mqtt.cpp#L87-L97). Is it possible to make it work for major version 2?

xwox, 2021-11-15:
Hello,
Does anyone know a work around?

mbehr1, 2021-11-16:
I think after a quick test/evaluation of the difference of version 1 to version 2 we can simply relax the check.
(can take a look on the weekend as well)

Matthias

mbehr1, 2021-11-20:
So the problem seems that the include file mosquitto.h defines LIBMOSQUITTO_MAJOR as 1 where as the mosquitto_lib_version reports it as 2.

mbehr1, 2021-11-20:
This looks like the include dir / included header version does not the match the version linked to.
@xwox, can you add your cmake output (should include "search for libmosquitto returned ...")? Thx.

xwox, 2021-11-20:
> This looks like the include dir / included header version does not the match the version linked to. @xwox, can you add your cmake output (should include "search for libmosquitto returned ...")? Thx.

I simply changed the make file so the version 2 is also accepted. everything seems to work.

mbehr1, 2021-11-20:
@xwox  so this can be closed, or?

Josef69, 2022-03-25:
@ngoldack could you find a solution for your problem? I have jet the same problem and hope you can help me.