volkszaehler/vzlogger #504: Bug in MeterExec debug logging?

sirtet, 2021-11-15:
I don't speak C++ but i think i found a bug...
Because i am not at all sure about what i see, i can't propose a fix, so just a report:

I was setting up an exec meter in vzlogger.conf and got confused from a logfile entry 
`[Nov 10 06:28:27][exec] MeterExec::open: Testing command line 'mbmd.py': Permission denied`
Permission denied seemed wrong, because the file did not even exist.

So, looking at the source, i think there's something wrong here, but i can't tell what exactly the intention behind the code is...
The output is produced by the `print` function on line 136 of [MeterExec.cpp](https://github.com/volkszaehler/vzlogger/blob/master/src/protocols/MeterExec.cpp)
```
	print(log_debug, "MeterExec::open: Testing command line '%s': %s", name().c_str(), command(),
		  strerror(errno));
	_pipe = popen(command(), "r");
```
But it seems to me the file is accessed only after the fact. So the error is from the last file operation before, which in my case is Permission denied... Am i right?

It seems, the error is again (now correctly) logged on line 141, so should there be only a static message `Testing command line...` on line 136, without the `%s` placeholders?
I see the different log-levels `log_debug` and `log_alert` being used. Still, can't tell how it is meant to work.

r00t-, 2021-11-15:
seems you are right.

(direct link to code for reference: https://github.com/volkszaehler/vzlogger/blob/ccb6d8ee49408cf6eb67b8d29492fcf7e82159d0/src/protocols/MeterExec.cpp#L136 )

actually, it's a little worse, the message is also misleading, as it's not "testing" anything, but actually executing the command.
(note that `popen()` will practically never fail, even if the program to be executed does not exist. - only `read()` will return EOF,and `pclose()` will later indicate the error.)

that has been in the MeterExec code ever since it was implemented:
https://github.com/volkszaehler/vzlogger/commit/4d19c75d42f21b9e0cecfbee6410e9538fe6f642#diff-d2a93ac7469329b680180bc63641d91571aeb6bdb4cd52f59dde44b2c89a3433R126

i guess that code simply didn't get enough review back then.

if you like, prepare an MR to fix this.
(you can just click "edit (pencil symbol)" in the github web interface.)

sirtet, 2021-11-16:
Oh, a fix is clearly outside my abilities. It was an unreasonable effort already to do the analysis as far as i got.

mbehr1, 2021-11-16:
> 4d19c75#diff-d2a93ac7469329b680180bc63641d91571aeb6bdb4cd52f59dde44b2c89a3433R126 <https://github.com/volkszaehler/vzlogger/commit/4d19c75d42f21b9e0cecfbee6410e9538fe6f642#diff-d2a93ac7469329b680180bc63641d91571aeb6bdb4cd52f59dde44b2c89a3433R126>

Hi,

yes, clearly wrong. I can fix it (but only next/coming weekend).

Matthias

mbehr1, 2021-11-20:
@sirtet, can you test the PR#506?

closed 2021-11-20 by mbehr1