volkszaehler/vzlogger #509: vzlogger has problems getting data from meters after reboot. USB meter replug (!) is necessary.

magicmil, 2022-01-09:
This startet around begin of November 2021.
Using Ubuntu 20.04.3 LTS always with newest updates.

After a reboot meter values are only comming very seldom or not at all.

The restart of the service does not suffice. Maybe it is even not necessary.
USB Replugging of the meteres is mandatory. Already updated to the newest version of vzlogger at the end of December 2021, which dit not help.
Afterwards it runs perfect until the next reboot - which is only needed it updates require it, since it is a 24/7 server.
I am using USB device, not port related configuration (see below)

Is it maybe related to a kernel change for the
usb-Silicon_Labs_CP2104_USB_to_UART_Bridge_Controller
my 2 meters are using?

Is it a known issue? Can I hard reset the controllers or USB config from commandline? I am wondering what a reboot could do less or mix up.

My volkszaehler and vzlogger setup had no problem for years since November 2021.
Here is my vzlogger.conf config:

/**
 * vzlogger configuration
 * 
 * use proper encoded JSON with javascript comments
 *
 * take a look at the wiki for detailed information:
 * http://wiki.volkszaehler.org/software/controller/vzlogger#configuration
 */

{
"retry" : 30,                   /* how long to sleep between failed requests, in seconds */
"daemon": true,                /* run periodically */
"verbosity" : 0,               /* between 0 and 15 */
"log" : "/var/log/vzlogger.log",/* path to logfile, optional */

"local" : {
        "enabled" : false,      /* should we start the local HTTPd for serving live readings? */
        "port" : 85,            /* the TCP port for the local HTTPd */
        "index" : true,         /* should we provide a index listing of available channels? */
        "timeout" : 30,         /* timeout for long polling comet requests, 0 disables comet, in seconds */
        "buffer" : 600          /* how long to buffer readings for the local interface, in seconds */
},

"meters" : [{
         "enabled" : true,
         "protocol" : "sml",
         "device" : "/dev/serial/by-id/usb-Silicon_Labs_CP2104_USB_to_UART_Bridge_Controller_00F3247A-if00-port0",
         "aggtime": 60,
         "aggfixedinterval": false,
         "channels": [{
             "uuid" : "7c087e70-9f71-11e5-a1ac-4bf832fe27de",
             "middleware" : "http://server:password@localhost:85/middleware.php",
             "aggmode": "MAX",
             "identifier" : "1-0:1.8.1", // Hausstrom Zaehlerstand (Wh)
             }, {
             "uuid" : "bcc9aca0-9f83-11e5-a195-d3a6a6707171", 
             "middleware" : "http://server:password@localhost:85/middleware.php",
             "aggmode": "AVG",  
             "identifier" : "1-0:1.7.1", // Hausstrom Leistung (Watt)
         }]
         },{
        "enabled" : true, 
        "protocol" : "sml",
        "device" : "/dev/serial/by-id/usb-Silicon_Labs_CP2104_USB_to_UART_Bridge_Controller_00F31C2A-if00-port0",
        "aggtime": 60,
        "aggfixedinterval": false,
        "channels": [{
             "uuid" : "d3273de0-a28c-11e5-8b21-f97c0c7dd262",
             "middleware" : "http://server:password@localhost:85/middleware.php",
             "aggmode": "MAX",
             "identifier" : "1-0:1.8.1", // Wärmepumpe Zaehlerstand (Wh)
             },{
             "uuid" : "a979cb30-a28c-11e5-b498-978c6272f141", 
             "middleware" : "http://server:password@localhost:85/middleware.php",
             "aggmode": "AVG",
             "identifier" : "1-0:1.7.1", // Wärmepumpe Leistung (Watt)
         }]
     }
]}

magicmil, 2022-01-11:
Seems to be fixed in 5.4.0-94-generic :-)

r00t-, 2022-04-04:
hard to imagine anything in vzlogger causing this.
a kernel/driver bug seems much more likely.
somr port power control is possible in linux, but i don't know if the raspberry pi's ports support it. maybe an extra hub would. (might add a pounter to details later.)
i'd suggest power-cycling (instead of just rebooting) on updates as a workaround, as this is something that can be planned.

gr8bit, 2022-11-06:
@magicmil I have the same issue and in my case it's vzlogger being unable to open the device when my Pi boots. When I restart vzlogger (sudo systemctl restart vzlogger) after booting, the device is there and vzlogger starts. Were you able to solve that problem?

magicmil, 2022-11-06:
Unfortunately not :(

gr8bit, 2022-11-06:
I fixed my problem by making the vzlogger service restart when it quits. This way, it will fail once on start because the /dev/ttyUSB0 isn't there and then becomes restarted after 30 seconds and succeeds because the device is there.

In file `/etc/systemd/system/vzlogger.service`, add `Restart` and `RestartSec` directives:
```
[Service]
...
Restart=always
RestartSec=30
```

Maybe that helps you as well.

r00t-, 2022-12-28:
@gr8bit:    
in your case you probably need some dependency in the systemd file to delay the start of vzlogger until the device is available... i wouldn't know how exactly, something with udev?