volkszaehler/vzlogger #510: InfluxDB - Write no data

etofi, 2022-01-09:
I use this [Docker Image](https://hub.docker.com/r/stefanschoof/vzlogger/) on my Rapberry Pi4 .
Unfortunately the data is not written to InfluxDB.

With this CURL call data arrives in the InfluxDB:
```
curl -i -XPOST 'http://192.168.178.59:8086/write?db=vzlogger' --data-binary 'vz_measurement value=0.64 1434055562000000000'

HTTP/1.1 204 No Content
Content-Type: application/json
Request-Id: 383eb51c-7185-11ec-87c9-0242ac110005
X-Influxdb-Build: OSS
X-Influxdb-Version: 1.8.6
X-Request-Id: 383eb51c-7185-11ec-87c9-0242ac110005
Date: Sun, 09 Jan 2022 19:49:13 GMT
```

Unfortunately, however, the Docker container does not write any data to my InfluxDB.
Have I set something wrong or is this a general problem?

This is my docker create:
```
sudo docker create \
	  --name=vzlogger \
	  --restart always \
	  --privileged \
	  --device /dev/ttyUSB0 \
	  -v /home/pi/docker/vzlogger/config/vzlogger.conf:/etc/vzlogger.conf \
	  -v /home/pi/docker/vzlogger/log:/var/log/ \
	  -e TZ="Europe/Berlin" \
	  stefanschoof/vzlogger:latest
```
This is my config
```
{
  "retry": 0,
  "daemon": true,
  "verbosity": 15,
  "log": "/var/log/vzlogger.log",
  "push": [],
  "local": {
    "enabled": false,
    "port": 8080,
    "index": false,
    "timeout": 0,
    "buffer": 0
  },
  "meters": [
    {
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
			"api": "influxdb",
			"uuid": "01234567-9abc-def0-1234-56789abcdefe",
			//"identifier": "1-0:1.8.0",
			"host": "192.168.178.59:8086",
			"database": "vzlogger",                        // Optional: make sure this database exists in InfluxDB
			"measurement_name": "vz_measurement",         // Optional: It is recommended that all your meters have the same InfluxDB measurement name
			"tags": "foo=bar,example=42",                 // Optional: Additional tags to append when inserting data
			//"username": "example",                        // Optional: When InfluxDB Auth is enabled you need to set the correct user and password
			//"password": "secure",
			//"token": "login token",                       // Optional: Use login token instead of username. Remove username from config to use token. 
			//"organization": "example.com",                // Optional: Configure organization to use /api/v2/write?org=xxx&bucket=database entry point
			//"max_batch_inserts": 4500,                    // Optional: Max number of measurements per request. No need to change this
			//"max_buffer_size": 450000,                    // Optional: Max number of measurements to be cached when InfluxDB is not available
			//"timeout": 30,                                // Optional: Time in seconds after which requests to InfluxDB time out
			"send_uuid": false,                           // Optional: Disables the sending of the UUID to the InfluxDB server
			"ssl_verifypeer": false,                      // Optional: Disables the certificate verification for https connections
        }
      ],
      "protocol": "random",
      "min": 0.0,
      "max": 40.0
    }
  ]
}
```

This is the log:
```
[Jan 09 20:43:22][main] vzlogger v0.8.0 based on HEAD-HASH-NOTFOUND from  ERR started.
[Jan 09 20:43:22][push] Ignoring push entry due to empty array or duplicate section
[Jan 09 20:43:22][mtr0] Creating new meter with protocol random.
[Jan 09 20:43:22][mtr0] Meter configured, enabled.
[Jan 09 20:43:22]       New meter initialized (protocol=random)
[Jan 09 20:43:22]       Configure channel.
[Jan 09 20:43:22]       Identifier is not set. Using default value 'NilIdentifier'.
[Jan 09 20:43:22][chn0] New channel initialized (uuid=...bcdefe api=influxdb id=NilIdentifier)
[Jan 09 20:43:22]       Have 1 meters.
[Jan 09 20:43:22][main] log level is 15
[Jan 09 20:43:22][main] local=0
[Jan 09 20:43:22]       Process not daemonized...
[Jan 09 20:43:22]       Opened logfile /var/log/vzlogger.log
[Jan 09 20:43:22][push] No pushDataServer defined.
[Jan 09 20:43:22][]     ===> Start meters
[Jan 09 20:43:22][mtr0] Meter connection established
[Jan 09 20:43:22][mtr0] Meter thread started
[Jan 09 20:43:22][mtr0] Meter is opened. Starting channels.
[Jan 09 20:43:22][mtr0] Number of readers: 1
[Jan 09 20:43:22][mtr0] Config.local: 0
[Jan 09 20:43:22][chn0] Start logging thread for influxdb-api.
[Jan 09 20:43:22][chn0] Logging thread started
[Jan 09 20:43:22][mtr0] Got 1 new readings from meter:
[Jan 09 20:43:22][chn0] InfluxDB API initialize
[Jan 09 20:43:22][]     Startup done.
[Jan 09 20:43:22][chn0] api InfluxDB using host 172.17.0.1:8086
[Jan 09 20:43:22][chn0] api InfluxDB no Token set
[Jan 09 20:43:22][mtr0] Reading: id=NilIdentifier/NilIdentifier value=19.01 ts=1641757402223
[Jan 09 20:43:22][chn0] api InfluxDB no organization set
[Jan 09 20:43:22][chn0] api InfluxDB no username set
[Jan 09 20:43:22][chn0] api InfluxDB no password set
[Jan 09 20:43:22][chn0] api InfluxDB using database vzlogger
[Jan 09 20:43:22][chn0] api InfluxDB using measurement name vz_measurement
[Jan 09 20:43:22][chn0] api InfluxDB using tags foo=bar,example=42
[Jan 09 20:43:22][chn0] api InfluxDB will use default timeout 30
[Jan 09 20:43:22][chn0] api InfluxDB will use default max_batch_inserts 4500
[Jan 09 20:43:22][chn0] api InfluxDB will use default max_buffer_size 450000
[Jan 09 20:43:22][chn0] api InfluxDB using send_uuid: false
[Jan 09 20:43:22][chn0] api InfluxDB using ssl_verifypeer: false
[Jan 09 20:43:22][chn0] api InfluxDB using url 172.17.0.1:8086/write?db=vzlogger&precision=ms
[Jan 09 20:43:22][chn0] Using InfluxDB api
[Jan 09 20:43:23][main] vzlogger v0.8.0 based on HEAD-HASH-NOTFOUND from  ERR started.
[Jan 09 20:43:23][push] Ignoring push entry due to empty array or duplicate section
[Jan 09 20:43:23][mtr0] Creating new meter with protocol random.
[Jan 09 20:43:23][mtr0] Meter configured, enabled.
[Jan 09 20:43:23]       New meter initialized (protocol=random)
[Jan 09 20:43:23]       Configure channel.
[Jan 09 20:43:23]       Identifier is not set. Using default value 'NilIdentifier'.
[Jan 09 20:43:23][chn0] New channel initialized (uuid=...bcdefe api=influxdb id=NilIdentifier)
[Jan 09 20:43:23]       Have 1 meters.
[Jan 09 20:43:23][main] log level is 15
[Jan 09 20:43:23][main] local=0
[Jan 09 20:43:23]       Process not daemonized...
[Jan 09 20:43:23]       Opened logfile /var/log/vzlogger.log
[Jan 09 20:43:23][push] No pushDataServer defined.
[Jan 09 20:43:23][]     ===> Start meters
[Jan 09 20:43:23][mtr0] Meter connection established
[Jan 09 20:43:23][mtr0] Meter thread started
[Jan 09 20:43:23][mtr0] Meter is opened. Starting channels.
[Jan 09 20:43:23][mtr0] Number of readers: 1
[Jan 09 20:43:23][chn0] Logging thread started
[Jan 09 20:43:23][chn0] Start logging thread for influxdb-api.
[Jan 09 20:43:23][]     Startup done.
```

Stefan-Code, 2022-01-09:
Based on the log file it seems like vzlogger is configured to use the wrong host `api InfluxDB using host 172.17.0.1:8086` instead of `192.168.178.59:8086`.

Also, vzlogger does not try to actually send data to InfluxDB. What you are seeing here is just initialization of the API. Is that the whole logfile? My guess is that there might be something wrong with the example config that you are using, but I can't really help with the configuration settings.

Nonetheless, can you also provide some additional information on how the network setup looks like? Specifically, which host/container is running InfluxDB and vzlogger? Also, are you running the curl example from inside the vzlogger container? If not, try to get a shell with `docker exec -it vzlogger /bin/sh` and try from there.

etofi, 2022-01-09:
The host is a raspberry 4 using raspios-bullseye-arm64-lite.
The InfluxDb is inside the docker image of Solaranzeige.
I can run the CURL call from the host and from any other docker containers.

It look like the vzlogger contaier crashes after 
`[Jan 09 20:43:23][chn0] Start logging thread for influxdb-api.`
Then it restarts.

So for the vzlogger container, I can't run the command.
`docker exec -it vzlogger /bin/sh. `
I always get the message
`Error response from daemon: Container 33a4df2e8df647d0288c68246ab3a8b43a89f1205deeb62b35fd55ff6d8ec73e is not running`

> Based on the log file it seems like vzlogger is configured to use the wrong host `api InfluxDB using host 172.17.0.1:8086` instead of `192.168.178.59:8086`.
I tried this with the same result.

etofi, 2022-01-10:
> I use this [Docker Image](https://hub.docker.com/r/stefanschoof/vzlogger/) on my Rapberry Pi4 .

Solved myself. The docker image is the problem. I buils my own image and now it works.

closed 2022-01-10 by etofi