volkszaehler/vzlogger #512: Docker - Dockerfile creates random volume

isarrider, 2022-01-19:
Hi,

the Dockerfile creates an empty additional volume...
I removed line 70 (and 69) to solved it...
I run it this way:

docker volume create vzlogger
(to create the volume where the /etc is mapped from the host)

docker run -d --restart unless-stopped \
 -p 11111:11111 \
 -v vzlogger:/etc \
 --device=/dev/ttyAMA0 \
 --name vzlogger \
 vzlogger
(11111 is my port for the webserver)...

Shall I do a PR?

BR,
Alex

r00t-, 2022-02-18:
```
# Setup volume
VOLUME ["/cfg"]
```
https://github.com/volkszaehler/vzlogger/blob/master/Dockerfile#L70

isarrider, 2022-02-18:
I prefer to have volumes by name, where I can the put the needed files in...
(the example above has of course a "docker volume create vzlogger" before,
then it is not named random)

m-reuter, 2022-02-20:
I agree that the volume can be removed. In my original version it was the hardcoded location of the config file, but with the two stage approach fddf47998843ad5e40b210b054ee5f6cce81f823 this changed.

isarrider, 2022-02-20:
I created a PR...

isarrider, 2022-03-09:
can be closed when PR #517 is in?

J-A-U, 2022-04-20:
Just merged

closed 2022-04-20 by J-A-U