volkszaehler/vzlogger #513: possible memleak...

isarrider, 2022-01-26:
Hi,
running the latest vzlogger build with all defaults
except #512 (removing line 69 & 70),
vzlogger runs and writes data to influx...
But RAM usage is slowly rising over time.
I will run it with a higher verbose level to see,
atm the config is:

[config.txt](https://github.com/volkszaehler/vzlogger/files/7941668/config.txt)

and the log is:
[Jan 25 21:12:28][main] vzlogger v0.8.1 based on heads/master-0-g037ab6b6a6-dirty from Sat, 20 Nov 2021 14:46:15 +0100 started.
[Jan 25 21:12:28][main] log level is 3

Ideas?

isarrider, 2022-01-26:
for completeness how I run it:
docker run -d --restart unless-stopped \
 -p 11111:11111 \
 -v vzlogger:/etc \
 --device=/dev/ttyAMA0 \
 --name vzlogger \
 -h vzlogger \
 vzlogger

isarrider, 2022-01-26:
6 hours after start, it is already at 47MB

isarrider, 2022-01-26:
[image]
this is the tendency...

isarrider, 2022-01-26:
I also mounted now /cfg, as it is written in the readme...
(although I also had to mount /etc, as it throws an error otherwise...)

isarrider, 2022-01-26:
[vzlogger_30.log](https://github.com/volkszaehler/vzlogger/files/7945210/vzlogger_30.log)
here is a log level 30 log...

isarrider, 2022-01-27:
Hi - thanks f your answer...
atm I am using just htop or der docker stats (or portainer)
but I can also give you
cat /proc/25237/status
or
pmap -x 25237
output (25237 is the current PID)...

I was also using ony my old Pi instance the vzlogger version from roughly mid Feb 2021
and it ran stable for month... (and the config file is identical except without the demon line)

isarrider, 2022-01-27:
let me come back to you on the gcore file...

isarrider, 2022-01-27:
[cat-status-1323.txt](https://github.com/volkszaehler/vzlogger/files/7950511/cat-status-1323.txt)
[cat-status-1349.txt](https://github.com/volkszaehler/vzlogger/files/7950512/cat-status-1349.txt)
cat/PID/status throws this...

J-A-U, 2022-01-27:
For comparison, my vzlogger uses a little over 80MB (with SML).
 based on git version: heads/master-0-g56d88a45be

isarrider, 2022-01-27:
> For comparison, my vzlogger uses a little over 80MB (with SML). based on git version: heads/master-0-g56d88a45be
right from the start? what ISA? stable then?

r00t-, 2022-01-27:
doing a little testing with your config and this construct,
adjuting the serial device to "/dev/stdin":
```
[~/vzlogger]$ while : ; do cat ../src/libsml-testing/*bin ; done | ./src/vzlogger -c ./vzlogger_issues_513.conf.local -f -o /dev/stderr 
```
if i set the api to "null",
memory is constant at 13mb,
if i set the api to influx with an invalid server, it seems to be growing at rate comparable to what you are seeing (given that the construct is processing readings at a MUCH higher rate than what your meter sends).
(i actually ran that test expecting memory to be filled with an influxdb "send buffer", but the code seems to be discarding the data if submitting fails?!)

so this indicates that the influxdb client might be leaking memory.

isarrider, 2022-01-27:
hmm - interesting...
but my influx writes are successful,
(see my log attached above...)

r00t-, 2022-01-27:
it might be leaking on both successful and unsuccessful writes.
maybe you can make a test with the api set to null, just to confirm that that "fixes" the leakage...
(i imagine you can't, because you would lose data.)
will have a look at the code and hunt for possible leaks.

r00t-, 2022-01-27:
found one leak here:
https://github.com/volkszaehler/vzlogger/commit/6ab1bb7d9c9e3e3d6417bef7ef61aa5117d66e56#r64739671

r00t-, 2022-01-27:
please test #514

isarrider, 2022-01-27:
nice...
(ah - I prob forgot to mention that I use the token feature to auth against Influx
but it becomes clear from the log...
still cant believe I am the only one who has this leak)...

So "before you call curl_slist_free_all on the list" is needed after, correct?

isarrider, 2022-01-27:
> please test #514

I will, thx... didnt see ur post 40min ago...
Yap, will do later tonight...

r00t-, 2022-01-27:
i broke it :shrug: ...
try again now

isarrider, 2022-01-27:
excatly the same...

[Jan 27 18:26:54][chn0] InfluxDB response was {"code":"unauthorized","message":"unauthorized access"}
[Jan 27 18:26:54][chn4] InfluxDB response was {"code":"unauthorized","message":"unauthorized access"}
[Jan 27 18:26:54][chn3] InfluxDB error! - HTTP Status 401
[Jan 27 18:26:54][chn3] InfluxDB response was {"code":"unauthorized","message":"unauthorized access"}
[Jan 27 18:26:54][chn8] InfluxDB error! - HTTP Status 401
[Jan 27 18:26:54][chn8] InfluxDB response was {"code":"unauthorized","message":"unauthorized access"}
[Jan 27 18:26:54][chn1] InfluxDB error! - HTTP Status 401
[Jan 27 18:26:54][chn1] InfluxDB response was {"code":"unauthorized","message":"unauthorized access"}
[Jan 27 18:26:54][chn7] InfluxDB error! - HTTP Status 401
[Jan 27 18:26:54][chn7] InfluxDB response was {"code":"unauthorized","message":"unauthorized access"}
[Jan 27 18:26:55][chn6] InfluxDB error! - HTTP Status 401
[Jan 27 18:26:55][chn6] InfluxDB response was {"code":"unauthorized","message":"unauthorized access"}
[Jan 27 18:26:55][chn2] InfluxDB error! - HTTP Status 401
[Jan 27 18:26:55][chn4] InfluxDB error! - HTTP Status 401
[Jan 27 18:26:55][chn4] InfluxDB response was {"code":"unauthorized","message":"unauthorized access"}
[Jan 27 18:26:55][chn3] InfluxDB error! - HTTP Status 401
[Jan 27 18:26:55][chn3] InfluxDB response was {"code":"unauthorized","message":"unauthorized access"}
[Jan 27 18:26:55][chn2] InfluxDB response was {"code":"unauthorized","message":"unauthorized access"}
[Jan 27 18:26:55][chn1] InfluxDB error! - HTTP Status 401

Waldmeisda, 2022-01-27:
Hi, thanks for the report, you are right, there is a issue 
[image]
I didn't spot this before my instance growed to 1.2G within 3 months.
I'll try to verify the fix as well.

isarrider, 2022-01-27:
@Waldmeisda 

that would be nice, as on my Pi4 the docker build takes ~20 min... ;)

Waldmeisda, 2022-01-27:
The Token is missing with your build https://github.com/volkszaehler/vzlogger/pull/514

I'll check whats wrong.

r00t-, 2022-01-27:
sorry for the breakage, will check

r00t-, 2022-01-27:
> that would be nice, as on my Pi4 the docker build takes ~20 min... ;)

unless there are changes in headers, only one file needs to be recompiled, which does not take as long

r00t-, 2022-01-27:
stupid mistake, try now...
```-               } else if (!_token_header) {
+               } else if (_token_header) {
```

Waldmeisda, 2022-01-27:
seams to work again, I did:
 2053  git branch -D MASTER
 2054  git fetch origin pull/514/head:MASTER
 2055  git switch MASTER
 2056  git log
 2058  git show
 2059  ./install.sh
 2060  sudo systemctl stop vzlogger
 2061  sudo systemctl start vzlogger
 2062  less /var/log/vzlogger.log
 2063  sudo tcpdump -nni any port 8086
No Errors, but I cant comment on memory yet.

isarrider, 2022-01-27:
> > that would be nice, as on my Pi4 the docker build takes ~20 min... ;)
> 
> unless there are changes in headers, only one file needs to be recompiled, which does not take as long

yap, but I didnt find a way with docker build to use the $ when the Hash is different
(even only from on char)
building a new container now...

isarrider, 2022-01-27:
looks good, you are the man, @r00t-, thank you...

[Jan 27 21:50:03][main] vzlogger v0.8.1 based on heads/master-0-g037ab6b6a6-dirty from Sat, 20 Nov 2021 14:46:15 +0100 started.
[Jan 27 21:50:03][main] log level is 3

log looks fine on level 3,
attached the portainer stats...
[image]

isarrider, 2022-01-27:
(another learning here for me was that the 32bit version is using ~ 0,5% more CPU Resources ;) )...
-> Screenshot from this morning bf the fix)...
[image]

isarrider, 2022-01-28:
fyi - still smooth at 4,3MB Ram like in the screenshot above!

Waldmeisda, 2022-02-17:
> I think we can close this...?

systemctl status vzlogger
● vzlogger.service - vzlogger
     Loaded: loaded (/etc/systemd/system/vzlogger.service; disabled; vendor preset: enabled)
     Active: active (running) since Thu 2022-01-27 21:34:13 CET; 2 weeks 6 days ago
   Main PID: 1652072 (vzlogger)
      Tasks: 5 (limit: 9324)
     Memory: 3.7M
     CGroup: /system.slice/vzlogger.service
             └─1652072 /usr/local/bin/vzlogger -c /etc/vzlogger.conf

--> still 3.7M after ~3 weeks. looks good. can be closed.

r00t-, 2022-02-17:
why did we close the ticket without merging the fix?

PR #514 (2022-02-17): [api/influxdb] fix memory leak (fixup 6ab1bb7d)
also create header and header-list only once.
(previous code leaked one curl_slist struct per send() invocation)

probably fixes #513
closed 2022-02-17 by r00t-