volkszaehler/vzlogger #521: Vzlogger only reads first 32 lines of data file

drtorchwood, 2022-05-09:
Hi, 
I use an external python script to get weather data from the DWD. I download the data once per day with 10 min resolution. 
Now, I want to read the file with vzlogger. In principle, the configuration is working but it only reads the first 32 lines. 

This is the part of the config:
```
    {
      "enabled": true,
      "allowskip": true,
      "interval": 600,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
          "api": "volkszaehler",
          "uuid": "f7fa6f00-cee8-11ec-ba4a-579e870e7bf9",
          "identifier": "radiation_global",
          "middleware": "http://localhost/middleware.php",
          "aggmode": "none",
          "duplicates": 0
        },
        {
          "api": "volkszaehler",
          "uuid": "29062040-cee9-11ec-a869-8741266b2e40",
          "identifier": "temperature_dew_point_mean_200",
          "middleware": "http://localhost/middleware.php",
          "aggmode": "none",
          "duplicates": 0
        }
      ],
      "protocol": "file",
      "path": "/home/pi/solar/dwd_daten.dat",
      "format": "$t $i $v"
    }
```

This is the log file. Because it reads the first 32 lines again, they are duplicates now. In the first run they have been added. If I delete the first lines from the file, the next ones are added. 
```
[May 10 00:21:41][main] vzlogger v0.8.1 based on heads/master-0-g76e868feed from Wed, 20 Apr 2022 08:39:54 +0200 started.
[May 10 00:21:41][push] Ignoring push entry due to empty array or duplicate section
[May 10 00:21:41][mtr0] Creating new meter with protocol exec.
[May 10 00:21:41][exec] MeterExec::MeterExec: Parsed format string "$t $i $v" => "%3$lf %2$ms %1$lf"
[May 10 00:21:41][mtr0] Meter configured, disabled.
[May 10 00:21:41]       New meter initialized (protocol=exec)
[May 10 00:21:41]       Configure channel.
[May 10 00:21:41][chn0] New channel initialized (uuid=...d81f4c api=volkszaehler id=Altitude)
[May 10 00:21:41]       Configure channel.
[May 10 00:21:41][chn1] New channel initialized (uuid=...2c38c5 api=volkszaehler id=Azimut)
[May 10 00:21:41]       Configure channel.
[May 10 00:21:41][chn2] New channel initialized (uuid=...688e1b api=volkszaehler id=Radiation)
[May 10 00:21:41][mtr1] Creating new meter with protocol file.
[May 10 00:21:41][file] Parsed format string "$t $i $v" => "%3$lf %2$ms %1$lf"
[May 10 00:21:41][mtr1] Meter configured, enabled.
[May 10 00:21:41]       New meter initialized (protocol=file)
[May 10 00:21:41]       Configure channel.
[May 10 00:21:41][chn3] New channel initialized (uuid=...0e7bf9 api=volkszaehler id=radiation_global)
[May 10 00:21:41]       Configure channel.
[May 10 00:21:41][chn4] New channel initialized (uuid=...6b2e40 api=volkszaehler id=temperature_dew_point_mean_200)
[May 10 00:21:41]       Have 2 meters.
[May 10 00:21:41][main] log level is 10
[May 10 00:21:41][main] local=1
[May 10 00:21:41]       Daemonize process...
[May 10 00:21:41]       Opened logfile /var/log/vzlogger.log
[May 10 00:21:41][]     ===> Start meters
[May 10 00:21:41][mtr0] Meter for protocol 'exec' is disabled. Skipping.
[May 10 00:21:41][mtr1] Meter connection established
[May 10 00:21:41][mtr1] Meter thread started
[May 10 00:21:41][mtr1] Meter is opened. Starting channels.
[May 10 00:21:41][chn3] Logging thread started
[May 10 00:21:41][chn4] Logging thread started
[... 125 lines cut ...]
[May 10 00:21:41][chn3] compare: 1651726200000 1651726800000
[May 10 00:21:41][chn3] compare: 1651726800000 1651727400000
[May 10 00:21:41][chn3] compare: 1651727400000 1651728000000
[May 10 00:21:41][chn3] compare: 1651728000000 1651728600000
[May 10 00:21:41][chn3] compare: 1651728600000 1651729200000
[May 10 00:21:41][chn3] compare: 1651729200000 1651729800000
[May 10 00:21:41][chn3] compare: 1651729800000 1651730400000
[May 10 00:21:41][chn3] compare: 1651730400000 1651731000000
[May 10 00:21:41][chn3] compare: 1651731000000 1651731600000
[May 10 00:21:41][chn3] compare: 1651731600000 1651732200000
[May 10 00:21:41][chn3] compare: 1651732200000 1651732800000
[May 10 00:21:41][chn3] compare: 1651732800000 1651733400000
[May 10 00:21:41][chn3] compare: 1651733400000 1651734000000
[May 10 00:21:41][chn3] compare: 1651734000000 1651734600000
[May 10 00:21:41][chn3] compare: 1651734600000 1651735200000
[May 10 00:21:41][chn3] compare: 1651735200000 1651735800000
[May 10 00:21:41][chn3] JSON request body: [ [ 1651717200000, 0 ], [ 1651717800000, 0 ], [ 1651718400000, 0 ], [ 1651719000000, 0 ], [ 1651719600000, 0 ], [ 1651720200000, 0 ], [ 1651720800000, 0 ], [ 1651721400000, 0 ], [ 1651722000000, 1000 ], [ 1651722600000, 3000 ], [ 1651723200000, 8000 ], [ 1651723800000, 15000 ], [ 1651724400000, 21000 ], [ 1651725000000, 27000 ], [ 1651725600000, 40000 ], [ 1651726200000, 55000 ], [ 1651726800000, 70000 ], [ 1651727400000, 58000 ], [ 1651728000000, 95000 ], [ 1651728600000, 127000 ], [ 1651729200000, 101000 ], [ 1651729800000, 122000 ], [ 1651730400000, 158000 ], [ 1651731000000, 172000 ], [ 1651731600000, 215000 ], [ 1651732200000, 210000 ], [ 1651732800000, 191000 ], [ 1651733400000, 199000 ], [ 1651734000000, 170000 ], [ 1651734600000, 181000 ], [ 1651735200000, 207000 ], [ 1651735800000, 208000 ] ]
[May 10 00:21:41][chn3] Middleware says duplicated value. Removing first entry!
[May 10 00:21:41][chn3] CURL Error from middleware: 'UniqueConstraintViolationException': 'An exception occurred while executing 'INSERT INTO data (channel_id, timestamp, value) VALUES (4,?,?), (4,?,?), (4,?,?), (4,?,?), (4,?,?), (4,?,?), (4,?,?), (4,?,?), (4,?,?), (4,?,?), (4,?,?), (4,?,?), (4,?,?), (4,?
[May 10 00:21:41][chn3] Waiting 0 secs for next request due to previous failure
```

Is there are way to make vzlogger read the whole file? I don't want to execute the python script many time because it takes some time (which is mostly overhead and not dependent on the amount of data that I download).

Thanks!

drtorchwood, 2022-05-16:
I had it running now for a few days. It turned out that it is skipping the one further line with every call until the end of the file is reached. However, it never starts reading from the beginning again after reading the whole file.

r00t-, 2022-06-10:
looking at the source:
https://github.com/volkszaehler/vzlogger/blob/76e868feede73b3f86b63d031768f71a227a903c/src/protocols/MeterFile.cpp#L239
https://github.com/volkszaehler/vzlogger/blob/76e868feede73b3f86b63d031768f71a227a903c/src/threads.cpp#L78
https://github.com/volkszaehler/vzlogger/blob/76e868feede73b3f86b63d031768f71a227a903c/src/Meter.cpp#L58
note that n=max_readings=32, the number on the line:
```
METER_DETAIL(file, File, "Read from file or fifo", 32),
```

the vzlogger core accepts a fixed maximum of readings from a meter per call to meter->read(),
which is set to 32 for MeterFile.
you could increase that and recompile,
but note that MeterFile was never intended to import data dumps,
it ahould probably be adjusted to remember the position in the file and return further readings in the next meter->read() call,
instead of resetting to the beginning of the file.
it should probably at least log a wsrning if max_readings is exceeded.