volkszaehler/vzlogger #533: Vzlogger struggles when Meter shifts from 3 byte to 2 byte in SML

fhwe71, 2022-07-13:
I have a DZG meter DWSB20.2TH to measure my power consumption and photovoltaic delivery.

Normally the meter sends 3 bytes with a factor 10^-2, e.g. 54 fe 9a 84 -> (ff ff ff - fe 9a 84 + 1)*0,01 = -915,15 Watt consumption, thus 915,15 Watt are fed into the grid. 

When the power consumption goes towards zero, the meter shifts to 2 bytes.
Example 53 80 f7 -> ( ffff-80 f7+1)*0,01 = -325,21 Watt consumption, 325,21W are delivered.

Nevertheless vzlogger assumes it's still 3 byte, and as the value is smaller than 7fffff, it calculates (0x80f7)*0,01 = 330,15 Watt consumption - which is a difference of more than 650 Watt!

This happens in the morning and evening when consumption is around zero. In the night, when it's really less than zero or during day delivery is high, 3 bytes are sent and everything is fine. Photos you can find here: [https://www.photovoltaikforum.com/thread/177445-fehler-smart-meter-oder-wechselrichterproblem/?postID=2688274#post2690322](url)

I wonder whether this is specific to DZG meters or happens with all meters. Nevertheless vzlogger is quite useless for me like that.
Can this be corrected?

fhwe71, 2022-07-14:
Done. I am no Github expert, hope everything was uploaded correctly. I processed the dumps with sml_server and the same effect occurs. Everything fine when 3 bytes are sent, wrong interpretation with 2 bytes. 
I think the meter does everything correctly, it uses type "5" for signed bytes + correct length. Anyway smllib recognizes the structure and length correctly, just it does not pass the information of 2 byte length to the conversion routine.

r00t-, 2022-12-25:
@fhwe71:
i'n not sure if i'm not missing something here,
but i don't see how the issue could possibly be in vzlogger instead of libsml.
all vzlogger does with sml readings is pass them to `sml_value_to_double()`, and then use the converted value returned by that:
https://github.com/volkszaehler/vzlogger/blob/27eb8d1566ec493d9cf64d2b53676c846ebb6e35/src/protocols/MeterSML.cpp
which appears to do the conversion (or just return a converted value generated when parsing, i'm not so deep into the libsml code):
https://github.com/volkszaehler/libsml/blob/559ca1e3ff8de7645fc4372f632e42b64cec780f/sml/src/sml_value.c

r00t-, 2022-12-25:
are these the expected values or wrong ones?
```
[libsml]$ ./examples/sml_server - <../vzlogger/dzg_dwsb20_2th_2byte.bin | grep 16.7
1-0:16.7.0*255#344.72#W
1-0:16.7.0*255#346.08#W
1-0:16.7.0*255#349.54#W
1-0:16.7.0*255#349.00#W
1-0:16.7.0*255#352.71#W
1-0:16.7.0*255#353.43#W
1-0:16.7.0*255#353.59#W
1-0:16.7.0*255#344.48#W
1-0:16.7.0*255#353.41#W
1-0:16.7.0*255#345.07#W
1-0:16.7.0*255#343.57#W
1-0:16.7.0*255#347.96#W
1-0:16.7.0*255#349.92#W
1-0:16.7.0*255#346.95#W
1-0:16.7.0*255#348.48#W
```

fhwe71, 2022-12-26:
Thanks for asking. I expect it's a libsml problem then. Currently I only can tell what I reported above was what I measured and it was incorrect. I will do a check with libsml in January, currently my data and equipment are not available.

jmberg, 2023-11-15:
Have you guys seen this? https://github.com/volkszaehler/libsml/issues/132