volkszaehler/vzlogger #534: vzlogger Service wird nach x Stunden automatisch beendet

michaelhutter, 2022-07-26:
Hab auf einem Raspberry Pi **vzlogger** am laufen (mit volkszaehler-Anbindung).
Grundsätzlich läuft es. Jedoch wird der Dienst offenbar nach einer gewissen Zeit automatisch beendet.
sudo systemctl status vzlogger
liefert dabei u. a. die Ausgabe "**CPU: #min ##.###s**".
Bei jedem Aufruf dieser Status-Abfrage erhöht sich die Zeit immer weiter.
Auf jeden Fall wird der Dienst nach einer gewissen Laufzeit (manchmal Minuten, manchmal Stunden) beendet.

```
michael@raspy4:~ $ sudo systemctl status vzlogger
? vzlogger.service - vzlogger
     Loaded: loaded (/etc/systemd/system/vzlogger.service; disabled; vendor pre>
     Active: active (running) since Tue 2022-07-26 13:31:45 CEST; 8h ago
   Main PID: 9399 (vzlogger)
      Tasks: 13 (limit: 1598)
        CPU: 8min 55.301s
     CGroup: /system.slice/vzlogger.service
             ??9399 /usr/local/bin/vzlogger -c /etc/vzlogger.conf

```
Frage: Handelt es sich bei "CPU" um die Laufzeit des Dienstes? Sollte ja eigentlich nicht mehr als ein paar Millisekunden dauern...
Was könnte der Grund sein, dass der Dienst im Laufe der Zeit immer mehr CPU-Zeit (?) benötigt?

r00t-, 2022-07-26:
die anzeige dort ist die kummulative cpu-zeit ueber die gesamte laufzeit des  prozesses, die natuerlicherweise kontinuierlich steigt,
da einen zusammenhang zum beenden konstruieren zu wollen ist voellig sinnfrei.

schaue lieber mal in's logfile, was da zum zeitpunkt des beendens steht.
die version vpn vzlogger und die config mit anzugeben waere dann auch irgendwann von vorteil.

michaelhutter, 2022-08-01:
Danke für die Info über die CPU-Laufzeit. Klingt plausibel.
Trotzdem beendet sich der Dienst nach wie vor nach etlichen Stunden Laufzeit.

**systemctl status vzlogger** bringt dann nach einigen Stunden folgende Ausgabe:

```
michael@raspy4:~ $ sudo systemctl status vzlogger
? vzlogger.service - vzlogger
     Loaded: loaded (/etc/systemd/system/vzlogger.service; disabled; vendor preset: enabled)
     Active: failed (Result: signal) since Sun 2022-07-31 21:41:35 CEST; 16h ago
    Process: 25061 ExecStart=/usr/local/bin/vzlogger -c /etc/vzlogger.conf (code=killed, signal=SEGV)
   Main PID: 25061 (code=killed, signal=SEGV)
        CPU: 52min 59.494s

Jul 29 16:42:18 raspy4 systemd[1]: Started vzlogger.
Jul 31 21:41:35 raspy4 systemd[1]: vzlogger.service: Main process exited, code=killed, status=11/SEGV
Jul 31 21:41:35 raspy4 systemd[1]: vzlogger.service: Failed with result 'signal'.
Jul 31 21:41:35 raspy4 systemd[1]: vzlogger.service: Consumed 52min 59.494s CPU time.

```
Während der Dienst läuft tut er seine Aufgabe völlig korrekt.

Im Logfile steht leider nichts. Da ich 9 Meters jede Sekunde auslese würde das Logfile sehr groß werden und außerdem meine SD-Karte kaputt schreiben. Deshalb habe ich den Loglevel auf 4 gesetzt.
```
{
"retry" : 0, /* sleep between failed requests (seconds) */
"daemon": true, /* run as deamon*/
"verbosity" : 4, /* Loglevel between 0 (nothing) and 15 (highest) */
"log" : "/var/log/vzlogger.log",/* logfile path */

```
_Auf welchen Wert müsste ich **verbosity** setzen, damit ich eine sinnvolle Ausgabe erhalte?
Oder gibt es unabhängig vom Logfile irgendwelche Dinge, die ich mir anschauen könnte?_

Ich spiele schon mit dem Gedanken über einen cron-Job minütlich abzufragen, ob der Dienst noch läuft und ihn bei Bedarf wieder zu starten ;-)

mbehr1, 2022-08-01:
signal=SEGV ist ein Crash. Wäre gut, wenn du Coredump aktiviert könntest (und dann anhängen/verteilen).
(Suche danach mal in der Maillinglist. Hatten wir vor paar Jahren schon mal als Thread „wie generiert man einen Coredump“)

michaelhutter, 2022-08-07:
Von welcher Mailinglist redest Du?
Leider ist mir nicht klar, was ein Coredump genau ist und wie ich das erstellen kann. Ist das was vzlogger-spezifisches oder eine Funktionalität von Linux? Ich vermute Letzteres.

Der Befehl "ulimit" liefert bei mir "unlimited" zurück.
Der Befehl "sysctl kernel.core_pattern" liefert "kernel.core_pattern = core" zurück.

Heißt das, dass bereits alles so konfiguriert ist, dass bei mir Coredumps geschrieben werden?
Wie komme ich an das Coredump dann ran?

michaelhutter, 2022-08-08:
Still the same. I don't see any message that a coredump was written. :-(
Any ideas how I could catch this error?

```
michael@raspy4:~ $ ulimit -c
unlimited

michael@raspy4:~ $ ulimit
unlimited

michael@raspy4:~ $ sudo systemctl status vzlogger
? vzlogger.service - vzlogger
     Loaded: loaded (/etc/systemd/system/vzlogger.service; disabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-08-07 22:33:24 CEST; 10min ago
   Main PID: 27946 (vzlogger)
      Tasks: 13 (limit: 1598)
        CPU: 10.243s
     CGroup: /system.slice/vzlogger.service
             ??27946 /usr/local/bin/vzlogger -c /etc/vzlogger.conf

Aug 07 22:33:24 raspy4 systemd[1]: Started vzlogger.

michael@raspy4:~ $ sudo systemctl status vzlogger
? vzlogger.service - vzlogger
     Loaded: loaded (/etc/systemd/system/vzlogger.service; disabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-08-07 22:33:24 CEST; 1h 17min ago
   Main PID: 27946 (vzlogger)
      Tasks: 13 (limit: 1598)
        CPU: 1min 17.776s
     CGroup: /system.slice/vzlogger.service
             ??27946 /usr/local/bin/vzlogger -c /etc/vzlogger.conf

Aug 07 22:33:24 raspy4 systemd[1]: Started vzlogger.

michael@raspy4:~ $ sudo systemctl status vzlogger
? vzlogger.service - vzlogger
     Loaded: loaded (/etc/systemd/system/vzlogger.service; disabled; vendor preset: enabled)
     Active: failed (Result: signal) since Mon 2022-08-08 00:54:45 CEST; 14h ago
    Process: 27946 ExecStart=/usr/local/bin/vzlogger -c /etc/vzlogger.conf (code=killed, signal=SEGV)
   Main PID: 27946 (code=killed, signal=SEGV)
        CPU: 2min 22.083s

Aug 07 22:33:24 raspy4 systemd[1]: Started vzlogger.
Aug 08 00:54:45 raspy4 systemd[1]: vzlogger.service: Main process exited, code=killed, status=11/SEGV
Aug 08 00:54:45 raspy4 systemd[1]: vzlogger.service: Failed with result 'signal'.
Aug 08 00:54:45 raspy4 systemd[1]: vzlogger.service: Consumed 2min 22.083s CPU time.
michael@raspy4:~ $ 

```

J-A-U, 2022-08-08:
> Von welcher Mailinglist redest Du?

https://wiki.volkszaehler.org/contact#mailing-listen

Zugschlus, 2022-08-08:
> Immer noch das gleiche. Ich sehe keinerlei Meldung, dass ein coredump geschrieben worden wäre :-( Irgendwelche Ideen, wie ich dem Fehler auf die Spur kommen könnte?
> 
> michael@raspy4:~ $ sudo systemctl status vzlogger
> ? vzlogger.service - vzlogger
>      Loaded: loaded (/etc/systemd/system/vzlogger.service; disabled; vendor preset: enabled)
>      Active: active (running) since Sun 2022-08-07 22:33:24 CEST; 10min ago
>    Main PID: 27946 (vzlogger)
>       Tasks: 13 (limit: 1598)
>         CPU: 10.243s
>      CGroup: /system.slice/vzlogger.service
>              ??27946 /usr/local/bin/vzlogger -c /etc/vzlogger.conf

systemd hat eigene Maßnahmen um Coredumps zu verhindern/zu kontrollieren. Eventuell ist es am einfachsten, /usr/local/bin/vzlogger -c /etc/vzlogger.conf manuell in einer shell (notfalls in einem screen) als der richtige User aufzurufen und auf den Absturz zu warten.

michaelhutter, 2022-08-10:
```
michael@raspy4:~ $ sudo /usr/local/bin/vzlogger -f -c /etc/vzlogger.conf 
[Aug 08 21:20:47][main] vzlogger v0.8.1 based on heads/master-0-g76e868feed from Wed, 20 Apr 2022 08:39:54 +0200 started.
[Aug 08 21:20:47][main] log level is 4
Speicherzugriffsfehler
michael@raspy4:~ $ 

```
Ich hab vzlogger jetzt mal manuell in einer Shell gestartet. Lief diesmal immerhin ca. 35 Stunden durch bis zum Absturz von vzlogger.
Leider war keine Ausgabe dabei, die mir hier weiter hilft. Bin etwas ratlos.
Der nächste Schritt wäre für mich nun ein Cron-Job, der überprüft, ob vzlogger noch läuft und ihn bei Bedarf wieder startet. Das wäre eine reine Symptombekämpfung anstatt einer sauberen Ursachenanalyse.

r00t-, 2022-08-12:
@michaelhutter : 
```ulimit -c unlimited```
siehe https://lists.volkszaehler.org/pipermail/volkszaehler-users/2015-March/005848.html
und dann https://lists.volkszaehler.org/pipermail/volkszaehler-users/2015-March/005868.html
es steht jedem frei mal eine wiki-seite dazu anzulegen.

muffl0n, 2022-09-10:
Du könntest systemd einfach sagen, dass er den Dienst "on-failure" restarten soll: https://www.freedesktop.org/software/systemd/man/systemd.service.html#Restart=
Also einfach `Restart=on-failure` im Abschnitt `[Service]` aufnehmen. 

Auch das ist nur Symptombekämpfung, aber immerhin besser als ein Cronjob. :)

michaelhutter, 2022-09-11:
Ich habs jetzt so gelöst, dass ich den Dienst einfach manuell über ein Shell-Script starte. Falls sich vzlogger beendet, so wird er einfach wieder gestartet:

```
#!/bin/bash
me=`basename "$0"`
LogFile=${me/sh/log}
VzConfFile=/etc/vzlogger.conf
while true
do
  sudo /usr/local/bin/vzlogger -f -c $VzConfFile | tee -a $LogFile
  sleep 1
done
```
Eigentlich ist das Script noch etwas aufwändiger, da ich auch noch logge, wann bzw. wie oft sich vzlogger beendet hat.
In letzter Zeit gibt es bei mir aber nur noch wenige dieser Abstürze, obwohl ich nichts geändert habe. Aber besser so, als dass ich mich anschließend über die fehlenden Daten ärgern muss.

r00t-, 2022-09-19:
@michaelhutter:
baue doch `ulimit -c unlimited` in dein script ein, und schicke uns die corefiles...

michaelhutter, 2022-09-19:
@r00t- Wie Du [weiter oben im Thread](https://github.com/volkszaehler/vzlogger/issues/534#issuecomment-1208125562) entnehmen kannst habe ich das schon erfolglos versucht.
Ich habe trotz `ulimit -c unlimited` kein Dumpfile finden können. Also blieb mir nur die Symptombekämpfung.
Abgesehen davon läuft vzlogger nun schon seit 19 Tagen ohne Absturz, obwohl ich nichts geändert habe und vorher alle paar Stunden einen Absturz hatte.
Wobei ein Unterschied fällt mir doch ein: ich logge mittlerweile nicht mehr ganz so viele Daten.

r00t-, 2023-12-01:
closing as cannot reproduce.
(would still be interested in that corefile to be able to fix the crash.)

closed 2023-12-01 by r00t-