volkszaehler/vzlogger #535: VZLogger won't compile on Ubuntu 22.04.1 LTS

tkuehne71, 2022-08-15:
I have tried to compile vzlogger from the current github source on Ubuntu 22.04.1 LTS. The build fails with errors due to the code using deprecated OpenSSL calls, i.e.:

`/usr/src/vzlogger/src/api/MySmartGrid.cpp:559:45: error: ‘HMAC_CTX* HMAC_CTX_new()’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
  559 |         HMAC_CTX *hmacContext = HMAC_CTX_new();
      | `

BoxedBrain, 2022-08-20:
Just got the same error. Did you found a solution yet?

tkuehne71, 2022-08-20:
no. Solution for me was to go back to Ubuntu 20.04 LTS

BoxedBrain, 2022-08-21:
@tkuehne71 doing a git fetch to PR #530, then you can compile and install on Ubuntu 22.04.
Hopefully this will be merged soon.

maxberger, 2022-09-25:
Right now you need #530 , #541 , #528 for vzlogger to compile and for the tests to pass.
Or you can use my branch with all my fixes: https://github.com/maxberger/vzlogger/tree/all

r00t-, 2023-01-03:
worked around by #557
to be fixed by #530
duplicated by #543 and #544

r00t-, 2023-01-17:
should be fixed with #530 merged

r00t-, 2023-01-17:
thanks to @maxberger for contributing the fix,
sorry that this took so long to merge.

closed 2023-01-17 by r00t-