volkszaehler/vzlogger #543: Compiling issues on Debian Bullseye (11.5)

FoxWhiskey, 2022-09-30:
Hi everyone,

I tried to compile vzlogger and noticed several issues. Prior any compiling attempts I cloned, compiled and successfully installed `libsml` from the volkszaehler-repository as requested by the installation manual.
1. Similarly to #535, a clone of the master tree does not compile on Debian Bullseye (11.5). 
2. I followed @maxberger 's advice and cloned his fork instead. As I prefer to install software from *deb - packages rather than from install-scripts (uninstall issue), I continued with the _build debian packages_ instructions and found the following:
   - `apt build-dep ./` requires `sudo`
   - `apt build-dep ./` then prompts the user to install `libsml-dev` from the debian-repositories - which I declined
   - `debuild` returns with an error, concerning `Unmet build dependencies: libsml-dev (>= 0.1.1)` and aborts. I find this confusing, as the freshly installed version of `libsml` is 1.0
   - `debuild -p` generates the *.deb - packages, however the build process exits with an error at the end, complaining about _missing secret keys_
3. `sudo dpkg --install vzlogger_0.3.5_amd64.deb` fails with an error as `libsml1 is not installed`

Most likely I did a couple of things in wrong way. But this is how far I got, consulting the available information/documentation.
Any help appreciated!

maxberger, 2022-10-01:
Hi Bernie,

1) Agreed. This is why I patched this in the first place.

2a) apt requires sudo: Yes, apt always requires sudo. You are installing something on your system, so you need to be root
2b/c) you declined installing the -dev package. -dev packages are needed to compile software which depends on the library., this is why the compile step fails. Install the -dev package.
2d) missing keys: I am not sure about this issue. Could you please try again with the -dev package installed, and then see what happens? I think this one can be ignored, but not sure.
3) looks like you need to instal llibsml1: sudo apt install libsml1

hth!

FoxWhiskey, 2022-10-01:
Servus and thanks for the quick reply!

As a matter of fact, I had some spare time yesterday evening and managed to get _vzlogger_ compiled as a deb-package, that installs and runs as expected!!
I pretty much went for a trial-and-error approach and found that the source of the issue was mostly user-induced, in addition to  some ambiguous passages (IMHO!) in the installation manual...

So here's what I've done/found and sorry in advance, if my comments are probably a bit too formal...
* I assumed, that `sudo` was necessary, but in other paragraphs of the manual `sudo` is stated where required, but it is not in _build as debian package_.
* paragraph _building libsml_ states that the debian-repository for _libsml-**dev**_ is no longer maintained and it is necessary to build an updated version. It is unclear, what _libsml.git_ actually provides: libsml or libsml-dev...
* when running `sudo apt build-dep ./` in a later step, _build-dep_ returns dependencies to `libsml1` and `libsml-dev` (which I declined during my first try, assuming that manually compiling `libsml` would suffice).  
So, apparently _libsml.git_ builds `libsml-dev`, but _libsml**1**_ is required nonetheless. `libsml1` is also missing in the list of required debian-packages.
* `debuild` without the `-p` switch still returns the version conflict message (see above), but `debuild -p` actually does the job.
* even with `libsml1` installed, the errors concerning _missing keys_ are still present but do not prevent the building process.

If I'm incorrect with my assumptions concerning libsml* - packages, please shine some light on it... ;-)

Best regards,
-bernie

r00t-, 2022-11-23:
> So, apparently libsml.git builds libsml-dev, but libsml1 is required nonetheless. libsml1 is also missing in the list of required debian-packages.

i think this is natural and expected.
a -dev package will always depend on the library it provides headers for, so the library does not need to be installled explicitly,
and both packages are naturally built from the same source.

r00t-, 2023-01-03:
@FoxWhiskey:
can you please provide a link to the "debian packages instructions" you are refering to?

FoxWhiskey, 2023-01-03:
> @FoxWhiskey: can you please provide a link to the "debian packages instructions" you are refering to?

Done!
Check my initial post, I've updated it with a link.
On the referred page, scroll down to chapter _Build as Debian-paket_.

r00t-, 2023-01-03:
note that dokuwiki provides anchors: https://wiki.volkszaehler.org/software/controller/vzlogger/installation_en#build_by_hand

r00t-, 2023-01-03:
i'm afraid those instructions are hopelessly outdated.
specifically, the libsml package in debian is still based on the old dailab repository and won't have any of the improvements we added in the last five years.
(see long thread here: https://github.com/volkszaehler/libsml/issues/72 ).
so those instructions would have to be updated to first build the libsml package from source and install it. (this should ideally be easy to do.)
getting our version into debian is the subject of above thread, and seems to be stalled still.

FoxWhiskey, 2023-01-04:
Hmm. Probably we're approaching the source of my confusion :-)

The instructions clearly state to clone _libsml_ from github and build it from source (, which I did) ! However, an additional package called _libsml1_ is still needed to compile _vzlogger_ (see third bullet point in my second post). 

Basically, I'm wondering about the context of _libsml_, _libsml1_ and _libsml(1?)-dev_...
(being aware, that a _-dev_ package provides headers for the library itself)

narc-Ontakac2, 2023-01-14:
Untested debian packagages for a recent libsml can be found here http://www.heute-morgen.de/debian/packages/.