volkszaehler/vzlogger #544: VZlogger SML Negative Werte für OBIS 1-0:36.7.0 (OBIS-Kennzahl für momentane Wirkleistung in Phase L1)

chrischris616, 2022-10-05:
leider kommen bei mir für den OBIS-Identifier negative Werte an. OBIS-Kennzahl für momentane Wirkleistung in Phase L1

Hier ein Dump des Zählers: Im Anhang sind die CSV-Daten aus dem Backend. Auch im Logfile werden die Daten negativ interpretiert. Tatsächliche Leistung ist aber ca. 500 W.

1b 1b 1b 1b 01 01 01 01 76 05 8d 9f 9b 13 62 00
62 00 72 63 01 01 76 01 01 02 31 0b 0a 01 44 5a
47 00 02 fc 07 a7 72 62 01 65 06 8a 5f 94 62 02
63 89 5a 00 76 05 8e 9f 9b 13 62 00 62 00 72 63
07 01 77 01 0b 0a 01 44 5a 47 00 02 fc 07 a7 07
01 00 62 0a ff ff 72 62 01 65 06 8a 5f 94 74 77
07 01 00 60 32 01 01 01 72 62 01 62 00 62 00 52
00 04 44 5a 47 01 77 07 01 00 60 01 00 ff 01 72
62 01 62 00 62 00 52 00 0b 0a 01 44 5a 47 00 02
fc 07 a7 01 77 07 01 00 01 08 00 ff 64 04 01 04
72 62 01 62 00 62 1e 52 ff 65 05 f9 4a b8 01 77
07 01 00 24 07 00 ff 01 72 62 01 62 00 62 1b 52
fe 53 c6 0c 01 01 01 63 90 8e 00 76 05 8f 9f 9b
13 62 00 62 00 72 63 02 01 71 01 63 94 11 00 00
1b 1b 1b 1b 1a 01 7c c8 1b 1b 1b 1b 01 01 01 01
76 05 90 9f 9b 13 62 00 62 00 72 63 01 01 76 01
01 02 31 0b 0a 01 44 5a 47 00 02 fc 07 a7 72 62
01 65 06 8a 5f 95 62 02 63 a0 93 00 76 05 91 9f
9b 13 62 00 62 00 72 63 07 01 77 01 0b 0a 01 44
5a 47 00 02 fc 07 a7 07 01 00 62 0a ff ff 72 62
01 65 06 8a 5f 95 74 77 07 01 00 60 32 01 01 01
72 62 01 62 00 62 00 52 00 04 44 5a 47 01 77 07
01 00 60 01 00 ff 01 72 62 01 62 00 62 00 52 00
0b 0a 01 44 5a 47 00 02 fc 07 a7 01 77 07 01 00
01 08 00 ff 64 04 01 04 72 62 01 62 00 62 1e 52
ff 65 05 f9 4a ba 01 77 

07 01 00 24 07 00 ff 01
72 
62 01 
62 00 
62 1b 52 fe 53 c2 c6 01 01 01 63
be f0 00 76 05 92 9f 9b 13 62 00 62 00 72 63 02
01 71 01 63 73 a5 00 00 1b 1b 1b 1b 1a 01 f3 44
1b 1b 1b 1b 01 01 01 01 76 05 93 9f 9b 13 62 00
62 00 72 63 01 01 76 01 01 02 31 0b 0a 01 44 5a
47 00 02 fc 07 a7 72 62 01 65 06 8a 5f 96 62 02
63 8d 8b 00 76 05 94 9f 9b 13 62 00 62 00 72 63
07 01 
77 01 0b 0a 01 44 5a 47 00 02 fc 07 a7 07
01 00 62 0a ff ff 72 62 01 65 06 8a 5f 96 74 
77 07 01 00 60 32 01 01 01 72 62 01 62 00 62 00 52
00 04 44 5a 47 01 
77 07 01 00 60 01 00 ff 01 72
62 01 62 00 62 00 52 00 0b 0a 01 44 5a 47 00 02
fc 07 a7 01 
77 07 01 00 01 08 00 ff 64 04 01 04
72 62 01 62 00 62 1e 52 ff 65 05 f9 4a bb 01 
77 
07 01 00 24 07 00 ff 
01 
72 
62 01 unit 
62 00 scaler
62 1b 52 fe 53 c3 18 <== hier müsste der Wert übertragen werden. 0x62 bedeutet unsigned int!
01 

01 01 63 c8 d5 00 

76 05 95 9f 9b
13 62 00 62 00 72 63 02 01 71 01 63 f4 d5 00 00

Ausschnitt aus Logfile:

[Oct 02 10:07:42][chn1] Adding reading to queue (value=-172.45 ts=1664698062720)
[Oct 02 10:07:43][chn0] Adding reading to queue (value=10022357.90 ts=1664698063720)
[Oct 02 10:07:43][chn1] Adding reading to queue (value=-171.56 ts=1664698063720)
[Oct 02 10:07:44][chn0] Adding reading to queue (value=10022358.00 ts=1664698064721)
[Oct 02 10:07:44][chn1] Adding reading to queue (value=-177.93 ts=1664698064721)
[Oct 02 10:07:45][chn0] Adding reading to queue (value=10022358.20 ts=1664698065722)
[Oct 02 10:07:45][chn1] Adding reading to queue (value=-170.13 ts=1664698065722)
[Oct 02 10:07:46][chn0] Adding reading to queue (value=10022358.30 ts=1664698066722)
[Oct 02 10:07:46][chn1] Adding reading to queue (value=-169.82 ts=1664698066722)
[Oct 02 10:07:47][chn0] Adding reading to queue (value=10022358.40 ts=1664698067723)
[Oct 02 10:07:47][chn1] Adding reading to queue (value=-169.03 ts=1664698067723)
[Oct 02 10:07:48][chn0] Adding reading to queue (value=10022358.60 ts=1664698068723)
[Oct 02 10:07:48][chn1] Adding reading to queue (value=-169.22 ts=1664698068723)
[Oct 02 10:07:49][chn0] Adding reading to queue (value=10022358.70 ts=1664698069724)
[Oct 02 10:07:49][chn1] Adding reading to queue (value=-170.49 ts=1664698069724)

{
  "retry": 0,
  "verbosity": 1,
  "log": "/var/log/vzlogger/vzlogger.log",
  "push": [],
  "local": {
    "enabled": false,
    "port": 8080,
    "index": false,
    "timeout": 0,
    "buffer": 0
  },
  "meters": [
    {
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
          "api": "volkszaehler",
          "uuid": "170102e0-3b6b-11ed-ae42-2f91b72369de",
          "identifier": "1-0:1.8.0",
          "middleware": "zzz",
          "aggmode": "max",
          "duplicates": 0
        },
        {
          "api": "volkszaehler",
          "uuid": "zzz",
          "identifier": "1-0:36.7.0",
          "middleware": "https://demo.volkszaehler.org/middleware.php",
          "aggmode": "none",
          "duplicates": 0
        }
      ],
      "protocol": "sml",
      "device": "/dev/ttyUSB0",
      "pullseq": "",
      "baudrate": 9600,
      "parity": "8n1",
      "use_local_time": true
    }
  ]
}

Der Zähler ist ein WS7412.1

Habe ich irgendwas falsch konfiguriert?
Über eine kurze Rückmeldung wäre ich sehr dankbar.

Viele Grüße

r00t-, 2022-11-23:
der fehler (wenn es einer ist) ist hoechstwahrscheiich nicht in vzlogger sondern in [libsml](https://github.com/volkszaehler/libsml).
magst du versuchen, das verhalten mit dem `sml_servrr` beispielprogramm von dort zu reproduzieren, und dann dort als bug melden?

jmberg, 2024-01-06:
Ist das hier jetzt nicht ein Doppel von volkszaehler/libsml#125?

r00t-, 2024-01-07:
wie oben gesagt, ist hier eher libsml als vzlogger betroffen. also war es richtig https://github.com/volkszaehler/libsml/issues/125  zu oeffnen.

closed 2024-01-07 by r00t-