volkszaehler/vzlogger #547: Absolute decimal values from meter become unprecise when sent to mqtt

gr8bit, 2022-11-05:
Software: vzlogger v0.8.1 based on heads/master-0-g27eb8d1566 from Wed, 28 Sep 2022 16:27:45 +0200
Host: not important
Counter: not important
Device: IR D0

When reading absolute counter values from my energy meter, they are being transmitted correctly as the debug mode shows. When they're submitted to mqtt though, they seem to get the "single float inaccuracy treatment" and are sometimes sent with lots of digits after the comma:
[image]

Is there any configuration I can use to prevent this from happening? Or is it a bug somewhere? My C++ is considerably bad so I wasn't able to find the cause myself...

r00t-, 2022-12-24:
the debug output actually forces two digits of precision:
https://github.com/volkszaehler/vzlogger/blob/76e868feede73b3f86b63d031768f71a227a903c/src/threads.cpp#L87
the influxdb api forces six digits:
https://github.com/volkszaehler/vzlogger/blob/76e868feede73b3f86b63d031768f71a227a903c/src/api/InfluxDB.cpp#L301
mysmartgrid seems to only deal with integers:
https://github.com/volkszaehler/vzlogger/blob/76e868feede73b3f86b63d031768f71a227a903c/src/api/MySmartGrid.cpp#L525
only the volkszaehler api seems to not limit the precision (like mqtt):
https://github.com/volkszaehler/vzlogger/blob/76e868feede73b3f86b63d031768f71a227a903c/src/api/Volkszaehler.cpp#L242

r00t-, 2022-12-27:
i very much dislike the wording of the ticket title, the numbers are not "imprecise", rather, too much precision is used in the binary->decimal conversion.
the causes are descibed here: https://en.wikipedia.org/wiki/Floating-point_arithmetic#Representable_numbers,_conversion_and_rounding
but i'm still not sure what the correct term for the effect is. "quantization error" is my best guess so far.

r00t-, 2022-12-27:
note that a very related floating-point related issue is discussed here:
https://github.com/volkszaehler/vzlogger/issues/500