volkszaehler/vzlogger #548: Debuild failed on Raspberry running Version 11 bullseye

DerPitter, 2022-11-14:
Try to run debuild, described at https://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version

make[2]: Leaving directory '/home/peter/vzlogger/obj-arm-linux-gnueabihf'
make[1]: *** [Makefile:185: all] Error 2
make[1]: Leaving directory '/home/peter/vzlogger/obj-arm-linux-gnueabihf'
dh_auto_build: error: cd obj-arm-linux-gnueabihf && make -j4 "INSTALL=install --strip-program=true" VERBOSE=1 returned exit code 2
make: *** [debian/rules:13: build] Error 25
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -us -uc -ui failed
peter@raspberrypi:~/vzlogger $ uname -a
Linux raspberrypi 5.15.76-v7l+ #1597 SMP Fri Nov 4 12:14:58 GMT 2022 armv7l GNU/Linux
peter@raspberrypi:~/vzlogger $ cat /sys/firmware/devicetree/base/model
Raspberry Pi 4 Model B Rev 1.5peter@raspberrypi:~/vzlogger $
peter@raspberrypi:~/vzlogger $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
peter@raspberrypi:~/vzlogger $

Full CLI output:

[debuild_failure.txt](https://github.com/volkszaehler/vzlogger/files/10001345/debuild_failure.txt)

r00t-, 2022-11-23:
just FYI, you cut off the log just at the most interesting point.
make's purpose is to run other programs, and we are interested in the program invokation that failed, which will be just before the first occurence of `make: Error`.

r00t-, 2022-11-23:
sadly even then the error message isn't very helpful.
can you try to run make manually in the obj-arm... directory?
```
[ 29%] Linking CXX static library libvz.a
cd /home/peter/vzlogger/obj-arm-linux-gnueabihf/src && /usr/bin/cmake -P CMakeFiles/vz.dir/cmake_clean_target.cmake
cd /home/peter/vzlogger/obj-arm-linux-gnueabihf/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/vz.dir/link.txt --verbose=1
/usr/bin/ar qc libvz.a CMakeFiles/vz.dir/Channel.cpp.o CMakeFiles/vz.dir/Config_Options.cpp.o CMakeFiles/vz.dir/threads.cpp.o CMakeFiles/vz.dir/Buffer.cpp.o CMakeFiles/vz.dir/Obis.cpp.o CMakeFiles/vz.dir/Options.cpp.o CMakeFiles/vz.dir/Reading.cpp.o CMakeFiles/vz.dir/exception.cpp.o CMakeFiles/vz.dir/local.cpp.o CMakeFiles/vz.dir/mqtt.cpp.o CMakeFiles/vz.dir/MeterMap.cpp.o
/usr/bin/ranlib libvz.a
make[3]: Leaving directory '/home/peter/vzlogger/obj-arm-linux-gnueabihf'
[ 29%] Built target vz
make[2]: Leaving directory '/home/peter/vzlogger/obj-arm-linux-gnueabihf'
make[1]: *** [Makefile:185: all] Error 2
make[1]: Leaving directory '/home/peter/vzlogger/obj-arm-linux-gnueabihf'
dh_auto_build: error: cd obj-arm-linux-gnueabihf && make -j4 "INSTALL=install --strip-program=true" VERBOSE=1 returned exit code 2
make: *** [debian/rules:13: build] Error 25
```

r00t-, 2023-01-03:
should be fixed by https://github.com/volkszaehler/vzlogger/pull/541

closed 2023-01-03 by r00t-

DerPitter, 2023-01-13:
Sorry no time for testing before today.
Fresh installed Rasbian.

Run steps :
```
apt-get install build-essential git cmake pkg-config subversion libcurl4-openssl-dev libgnutls28-dev libsasl2-dev uuid-dev libtool libssl-dev libgcrypt20-dev libmicrohttpd-dev libltdl-dev libjson-c-dev libleptonica-dev libmosquitto-dev libunistring-dev dh-autoreconf
git clone https://github.com/volkszaehler/vzlogger.git
cd vzlogger
mkdir build
cd build
cmake -DBUILD_TEST=off ..
make
sudo make install
```

Compile läuft durch.
In den Tests kommt es zu einem Fehler :

```
/usr/bin/ctest --force-new-ctest-process -j4
Test project /home/peter/vzlogger/obj-arm-linux-gnueabihf
    Start 1: vzlogger_unit_tests
    Start 2: mock_metermap
    Start 3: mock_MeterW1therm
    Start 4: mock_MeterS0
1/4 Test #2: mock_metermap ....................Child aborted***Exception:   0.08 sec
[==========] Running 2 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 2 tests from mock_metermap
[ RUN      ] mock_metermap.basic_open_close_enabled
terminate called without an active exception

2/4 Test #3: mock_MeterW1therm ................   Passed    0.31 sec
3/4 Test #1: vzlogger_unit_tests ..............   Passed    0.69 sec
4/4 Test #4: mock_MeterS0 .....................   Passed    3.28 sec

75% tests passed, 1 tests failed out of 4

Total Test time (real) =   3.38 sec

The following tests FAILED:
          2 - mock_metermap (Child aborted)
Errors while running CTest
make[1]: *** [Makefile:128: test] Error 8
make[1]: Leaving directory '/home/peter/vzlogger/obj-arm-linux-gnueabihf'
dh_auto_test: error: cd obj-arm-linux-gnueabihf && make -j4 test ARGS\+=-j4 returned exit code 2
make: *** [debian/rules:13: binary] Error 25
dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -us -uc -ui failed

```

vzlogger/obj-arm-linux-gnueabihf/Testing/Temporary/LastTest.log

```
2/4 Testing: mock_metermap
2/4 Test: mock_metermap
Command: "/home/peter/vzlogger/obj-arm-linux-gnueabihf/tests/mocks/mock_metermap"
Directory: /home/peter/vzlogger/obj-arm-linux-gnueabihf/tests/mocks
"mock_metermap" start time: Jan 13 13:06 CET
Output:
----------------------------------------------------------
[==========] Running 2 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 2 tests from mock_metermap
[ RUN      ] mock_metermap.basic_open_close_enabled
terminate called without an active exception
<end of output>
Test time =   0.70 sec
----------------------------------------------------------
Test Failed.
"mock_metermap" end time: Jan 13 13:06 CET
"mock_metermap" time elapsed: 00:00:00
----------------------------------------------------------

```

Sorry, ich habe keinen Schimmer vom Coden, bin Anwender.

r00t-, 2023-01-13:
```
[ RUN      ] mock_metermap.basic_open_close_enabled
terminate called without an active exception
```

this will be fixed by https://github.com/volkszaehler/vzlogger/pull/528 once merged