volkszaehler/vzlogger #553: variables are used in this project, but they are set to NOTFOUND.

UteHaus, 2022-12-21:
Hey,

I get every time following message with the `install.sh`
I use Pi zero with:
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye

```
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBGCRYPT
    linked by target "vzlogger" in directory /home/adi/strom/vzlogger/src
OPENSSL_CRYPTO_LIBRARY (ADVANCED)
    linked by target "vzlogger" in directory /home/adi/strom/vzlogger/src
    linked by target "vzlogger_unit_tests" in directory /home/adi/strom/vzlogger/tests
    linked by target "mock_MeterOMS" in directory /home/adi/strom/vzlogger/tests/mocks
    linked by target "mock_metermap" in directory /home/adi/strom/vzlogger/tests/mocks
OPENSSL_SSL_LIBRARY (ADVANCED)
    linked by target "vzlogger" in directory /home/adi/strom/vzlogger/src
    linked by target "vzlogger_unit_tests" in directory /home/adi/strom/vzlogger/tests
    linked by target "mock_MeterOMS" in directory /home/adi/strom/vzlogger/tests/mocks
    linked by target "mock_metermap" in directory /home/adi/strom/vzlogger/tests/mocks

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

```

r00t-, 2022-12-21:
```apt-get install libssl-dev libgcrypt20-dev```?

r00t-, 2022-12-27:
i wonder if `install.sh` should check for those dependencies.

r00t-, 2023-01-17:
looks like this was reported before at #197