volkszaehler/vzlogger #554: MySmartGrid::hmac_sha1 gives warnings with OpenSSL 3

narc-Ontakac2, 2023-01-02:
I am currently trying to compile vzlogger on debian bookworm. This gives warnings like
/home/jo/projects/debian/vzlogger/src/api/MySmartGrid.cpp:559:45: error: ‘HMAC_CTX* HMAC_CTX_new()’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
  559 |         HMAC_CTX *hmacContext = HMAC_CTX_new();
      |                                 ~~~~~~~~~~~~^~
`
Taking a closer look I found that OpenSSL below 1.1 is still supported. Is this still needed? I can create a patch, but things would be easier if that could be dropped. 

See also
https://github.com/openssl/openssl/blob/master/doc/man7/migration_guide.pod#deprecated-low-level-mac-functions

narc-Ontakac2, 2023-01-02:
OK, after having fixed this myself I found PR 530. I just do not understand why it has not yet been merged, I am however aware that this is not necessarily simple. 

This prevents a build on a soon to be frozen debian testing, so it should happen soon.

r00t-, 2023-01-02:
duplicate of #535 etc.

r00t-, 2023-01-02:
@narc-Ontakac2:
it appears that #530 is blocked by #528 ,  which is kind of stalled.
yes, this is not good.

r00t-, 2023-01-03:
worked around by #557
to be fixed by #530
duplicate of #543 and #535

closed 2023-01-03 by r00t-