volkszaehler/vzlogger #558: Official vzlogger Debian package

narc-Ontakac2, 2023-01-04:
As those who read #331 may have guessed I intend to package an official vzlogger debian package. Since I am not a Debian developer I will need a sponsor to do reviews and uploads, I am currently looking for one.

One thing I need would be an official version that is easier to determine than finding the commit that changed the version in CMakeList.txt. This could be a tag or a github release.

r00t-, 2023-01-04:
can/will you also take care of the libsml debian package? (which is still build from the obsolete dailab source.)
https://github.com/volkszaehler/libsml/issues/72

narc-Ontakac2, 2023-01-04:
Yes, but I need the current maintainer @Ampelbein to allow that.

narc-Ontakac2, 2023-01-04:
Why is googletest downloaded via git? At first glance there is a debian package.

narc-Ontakac2, 2023-01-04:
Where do I find the source (MD?) for vzlogger_conf_parameter? This would make a quite useful man page.

r00t-, 2023-01-04:
> Why is googletest downloaded via git?

https://github.com/volkszaehler/vzlogger/commit/dc2161907fd8eea09364e1f5955199efd59c8fa3
@mbehr1 ?

mbehr1, 2023-01-05:
there was no specific reason. If you find a better way to install it, please go ahead :-)

narc-Ontakac2, 2023-01-05:
OK, there is PR #560 now. On Debian the necessary install is done with `apt install libgmock-dev libgtest-dev`. Adding that to the debian build dependencies will follow, debian/control is currently WiP.

narc-Ontakac2, 2023-01-06:
Notes on the state of packaging (mostly for myself):
[VZLogger ITP bug](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=864255)

narc-Ontakac2, 2023-03-12:
Again, mostly for myself
https://wiki.debian.org/PackageSalvaging
https://www.debian.org/doc/manuals/developers-reference/pkgs.en.html#package-salvaging

narc-Ontakac2, 2023-04-11:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1034250

narc-Ontakac2, 2024-03-20:
Vzlogger was uploaded to Debian unstable and is now in the [New Queue](https://ftp-master.debian.org/new.html). Unfortunately this will take some time. There will be a final review by the ftp masters, but I expect vzlogger to pass.

narc-Ontakac2, 2024-04-20:
[Done.](https://tracker.debian.org/pkg/vzlogger) The package information will take a few mor hours to appear.

closed 2024-04-20 by narc-Ontakac2

r00t-, 2024-04-22:
thanks for your work on this!

narc-Ontakac2, 2024-05-24:
Now also [on trixie](https://packages.debian.org/trixie/vzlogger).

justinotherguy, 2024-05-24:
> Now also [on trixie](https://packages.debian.org/trixie/vzlogger).

Champagner!!! :D 🍾 

Thx a million for your effort, @narc-Ontakac2 !