volkszaehler/vzlogger #561: Dockerfile has wrong parameters in final startup command

p4jo, 2023-01-05:
In the last line of the Dockerfile, it should be 
```
CMD ["vzlogger", "--foreground", "-c", "/cfg/vzlogger.conf"]
```
instead of
```
CMD ["vzlogger", "--foreground"]
```
because, else it doesn't work when simply following the instructions in `README.md`.

r00t-, 2023-01-05:
@p4jo:
no, that should not be necessary, as `/etc/vzlogger.conf` is a hardcoded default:
https://github.com/volkszaehler/vzlogger/blob/927c2634414b808797df4a1db6392c75516d6613/src/Config_Options.cpp#L43
which is overwritten when `-c` is specified:
https://github.com/volkszaehler/vzlogger/blob/927c2634414b808797df4a1db6392c75516d6613/src/vzlogger.cpp

r00t-, 2023-01-05:
refering to this content of README.md:
https://github.com/volkszaehler/vzlogger/blame/e6c98fda94d01271805209b9a2860c66099e3933/README.md#L62
added in https://github.com/volkszaehler/vzlogger/commit/e6c98fda94d01271805209b9a2860c66099e3933

i wonder if it makes sense to hardcode the `/cfg` path in the dockerifle,
as it wiill also have to be provided by the user in the invocation.

@m-reuter: can you provide your reasning on usage of the dockerfile?

syntetic81, 2023-10-30:
i run in this issue!
i get it working with:
`docker run --restart=always -v /home/pi/projects/vzlogger-docker:/etc`

r00t-, 2023-11-09:
i have no opinion on this, as i don't use docker and don't see who would,    
but overmounting /etc seems a litte more sensible than pointing vzlogger to /cfg.

m-reuter, 2023-11-09:
Hi, either is fine. It seems that vzlogger now looks for the config in /etc and maybe earlier versions looked for it in /cfg ? That would at least explain why it may have worked originally. 
Overwriting /etc may not be a great idea, because there are other files in /etc that would be missing and maybe there are mounted files that interfere with something. 

So I would prefer the other solution, always forcing users to mount to /cfg and then hard coding that path in the docker image , as /cfg is not used for anything else. But again, either approach should work.

r00t-, 2023-11-14:
we are building an image intended to only run vzlogger,
i think the only relevant file in /etc would be `resolv.conf` (as vzlogger uses the network),
and that is being injected by docker anyway.
just wonder if having a mountpoint at /etc breaks that mechanism.
```
[~]$ docker run -it -v testtest:/etc:nocopy debian:bookworm
bash-5.2# ls /etc
hostname  hosts  resolv.conf
```
docker writes the injected configs to the mounted volume, no idea if this is an issue.

r00t-, 2023-11-14:
> It seems that vzlogger now looks for the config in /etc and maybe earlier versions looked for it in /cfg ? That would at least explain why it may have worked originally.

even the plain C version >12 years ago had `/etc/vzlogger.conf`:
https://github.com/volkszaehler/vzlogger/commit/d125fd526c667d7cb6c0890e1f5511db12e9ed5a#diff-db6d72b498f4a0a0754cf614c6141dc0380706687e9ed5549b2041a8864c446eL310

i`m waiting for a user of the dockerfile to explain if it ever worked for them.

conmarti, 2023-12-18:
IMHO this could never have worked.
The solution I came up with is:
`docker run -d --restart=always -v /home/pi/projects/vzlogger-docker/vzlogger.conf:/etc/vzlogger.conf vzlogger`
which makes sure, that only the file vzlogger.conf is "bind mounted" into the container and everything else in /etc is left untouched.

closed 2026-04-25 by r00t-