volkszaehler/vzlogger #562: ensure timely exit of metering thread(s) with #528 merged

r00t-, 2023-01-15:
see discussion in #528

r00t-, 2023-01-27:
so we need to insert cancellation points in ALL meters...
(as indicated by testing of #572 (thanks @YARD2 ))

i wonder if we can get away with allowing cancellation around the call to meter->read() in the reading thread?
is there a real risk of a meter->read() function calling  a destructor?
@maxberger , @mbehr1  ...

maxberger, 2023-01-27:
The good news is that allowing cancellation around read() would (in the worst case) restore the behavior that was there before. So it is at most as risky as rolling back the original proposal.

The bad news is that it really depends what is in every read() function. I'll try to take a look at those this weekend, since I introduced the original bug I should probably help fixing it :)

r00t-, 2023-01-27:
the thread is only cancelled on shutdown, when we want to exit anyway, so imho any crash at that point is an issue of relatively low importance - thus it'd just allow cancellation around meter->read(). instead of adjusting every single meter.
(while the crash on shutdown IS an issue for the tests.)

maxberger, 2023-01-28:
The risk is that the meter puts the hardware in some strange state from which the hardware cannot recoder. In this case, it would be better if whatever hardware function is executed is completed before quitting.

If you want to, I've created PR #573 . I've looked at all meters for potential cancellation points. 

If you prefer making read() cancellable thats find too.

Interesting fact: Some of the meters seem to be waiting forever in read() , while some meters seem to quit immediately. Of course, the last ones don't need to be modified.

r00t-, 2023-03-08:
should be fixed by https://github.com/volkszaehler/vzlogger/pull/573

closed 2023-03-08 by r00t-