volkszaehler/vzlogger #568: Timestamps received from SML Meter are written to InfluxDB with wrong timezone assumption

ccwb-102, 2023-01-23:
If a SML meter supplies timestamp information those timestamps are used to write data to influxdb if configured as middleware. The problem is that the meter timestamps are in local time (at least for my meter) and influxdb expects UTC timestamps. This results in timestamps in influxdb that are off by the local UTC offset.

A solution could be to have a meter configuration parameter defining the timezone of the supplied timestamps and then to correct the timestamps accordingly before writing to the middleware. But this is without knowing anything about the internal vzlogger architecture.

J-A-U, 2023-01-23:
In cases your meter provide wrong timestamp use `"use_local_time": true,` within the meter-section.

[https://wiki.volkszaehler.org/software/controller/vzlogger/vzlogger_conf_parameter?s[]=use_local_time#sml](https://wiki.volkszaehler.org/software/controller/vzlogger/vzlogger_conf_parameter?s[]=use_local_time#sml)

r00t-, 2023-01-23:
@ccwb-102:
sadly use of timestaps without a timezone specification is extremely broken, in ways that we cannot fix.
does the meter implement daylight saving time? when the clock jumps backwards or forwards, things will get very messy.
(vzlogger would have to be perfectly in sync with the device's clock anyway to guess the timezone correctly, so you might just aswell `use_local_time` right away.)
could it be that the timezone is included in some SML field that we do not yet parse?
open for suggestions or MRs either way.

r00t-, 2023-01-23:
have you actually compared the behaviour of influxdb and the volkszaehler.org middleware?
because the volkszaehler api also uses unix timestamps:
https://github.com/volkszaehler/vzlogger/blob/929a8d65bf125ec35a1692711f101f03168864a3/src/api/Volkszaehler.cpp#L241
influx:
https://github.com/volkszaehler/vzlogger/blob/929a8d65bf125ec35a1692711f101f03168864a3/src/api/InfluxDB.cpp#L304

closed 2023-01-23 by J-A-U