volkszaehler/vzlogger #570: Use other Timestamp request

YARD2, 2023-01-24:
Hi,

I have the problem with Holley DTZ541.
The timestamp of 1.8.0 and 2.8.0 having the wrong timestamp, but the rest of the ID e.g. 16.7.0 has the right ones.
Would it be possible to check if the other values are ok, and use it also for the first ones ?
Currently I use "use_local_time": true
but I think using the right ones form the meter would be better.

[Jan 24 16:49:08][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=2167.50 ts=21441000
[Jan 24 16:49:08][mtr0] Reading: id=1-0:2.8.0*255/ObisIdentifier:1-0:2.8.0*255 value=0.00 ts=21441000
[Jan 24 16:49:08][mtr0] Reading: id=1-0:16.7.0*255/ObisIdentifier:1-0:16.7.0*255 value=310.00 ts=1674575348527

r00t-, 2023-01-24:
currently vzlogger uses each reading's timestamp for that reading:
https://github.com/volkszaehler/vzlogger/blob/929a8d65bf125ec35a1692711f101f03168864a3/src/protocols/MeterSML.cpp#L329

messing with this seems rather... messy...

some of the considerations made here apply:
https://github.com/volkszaehler/vzlogger/pull/476#pullrequestreview-627218150
(it's unfortunate that we parse the sml telegram entry by entry and create readings, so we don't have the data from other readings around (anymore or not yet), either to use data from them, or to manipulate their timestamps retroactively.)

YARD2, 2023-01-25:
Thx for explanation.
Then it would really be a bit complicated.