volkszaehler/vzlogger #571: Vzlogger can not stop

YARD2, 2023-01-25:
Hello,
system: Raspberry Pi, used latest vz image.
Initially it is working without problems.

Update to latest github version (https://wiki.volkszaehler.org/howto/raspberry_pi_image)
then vzlogger can not stop and hangs at the end

[Jan 25 10:51:23][push] Stopped push_data_thread

reboot will also be prevented.
Log level is 15, no more info provided from log
I can only use kill -9 [pid]

J-A-U, 2023-01-25:
Might be related to #528, we have an open Issue (as reminder) for this under #562 .

YARD2, 2023-01-25:
OK, then I wait.

r00t-, 2023-01-25:
@YARD2:
thanks for your report!
i do not know how often that image is rebuilt, and would be kind of surprised if it has that change already.
can you please confirm the exact version of vzlogger, a version number and git hash should be logged on startup.

J-A-U, 2023-01-25:
The Image is from July '22. But from context i assume @YARD2 did update vzlogger.

YARD2, 2023-01-25:
I used the image from July 22 (+lastest RasPi updates apt update & upgrade) and updated to the latest github verion of vzlogger
0.8.1
 based on git version: heads/master-0-g929a8d65bf
 last commit date: Thu, 19 Jan 2023 00:15:50 +0000
I could reproduce it with a RasPi zero, too.
Original image is working, after update vzlogger the problem occurs.

r00t-, 2023-01-26:
thanks for reporting!
will try to debug this as soon as i find the time (if nobody else beats me to it?).

r00t-, 2023-01-26:
closing this as a duplicate of #562

closed 2023-01-26 by r00t-

r00t-, 2023-01-26:
@YARD2:
can you test the totally untested and experimental fix in https://github.com/volkszaehler/vzlogger/pull/572 ?

justinotherguy, 2023-01-26:
Let me know, if I can assist ;)

YARD2, 2023-01-26:
Looks like that fix the issue.
I can start and stop the service and reobot without problems
Thanks !

r00t-, 2023-02-08:
should now be fixed in master by merging https://github.com/volkszaehler/vzlogger/pull/573
please do report any issues.